MKFirmware/MK4duo #336: Corexy homing Z in wrong direction

Dzikibiszkopt, 2017-12-07:
There is problem with homing z axis when using corexy mechanism. All wiring things are ok because when i just moving bed directions are ok (bed goes up when hit "-" direction on pronterface and goes down when "+" is triggered) but when i press homing button or give "g28 z" command bed goes to opposite direction (i changed even `Z_HOME_DIR` ) and problem is that bed z axis 0 is at bottom of printer and max at top - where printhead is.

It is tried at mk4duo and marlin, even fresh config with only steps per unit and mechanics changed. Bug is the same at both.

closed 2017-12-07 by Dzikibiszkopt

iosonopersia, 2017-12-07:
What? XD Did you solve your issue with z homing? Everything ok now?

Dzikibiszkopt, 2017-12-07:
yea changed title to not bother anyone... just my fckup with endstop config xD

iosonopersia, 2017-12-07:
That's good, don't worry xD