MKFirmware/MK4duo #337: Smart Ramps + Nextion HMI 4.3

jbaiao, 2017-12-07:
Hello.
I need help on the pinout that I need to connect NEXTION HMI 4.3 to a smart ramps.
It will be to install on a coreyx (VORON 1.5).
Thank you for your help

MagoKimbra, 2017-12-07:
On AUX 3 pin 1 and 2 is Pin D17 and D16 (RX2 and TX2 at Serial 2)
http://reprap.org/mediawiki/images/1/15/SMART_RAMPS_1-4-1_SCH_I2C.jpg
http://www.adrirobot.it/arduino/arduino_due/immagini/Due-pinout-A4_grande.png

jbaiao, 2017-12-07:
YOU ARE THE MAN!
Thank for the quick response.

jbaiao, 2017-12-07:
So .... in the wizard, in the LCD configuration, will I have to select the "Serial2" option in the "Nextion serial port" field?

MagoKimbra, 2017-12-07:
Yes!!!

jbaiao, 2017-12-11:
Tks!
Another question :)
To have sd card support, does nextion do this (micro sd slot) or do I have to use some adapter?

iosonopersia, 2017-12-11:
The micro-sd slot inside the nextion display is only used to update its internal firmware, thus its not readable by the Arduino. Yes, you'll have to add an external  micro-sd adapter (they're very cheap, search them online)

jbaiao, 2017-12-11:
And do you can tell me in AUX2 and AUX what is the pins to connect:
5V
SCK
MISO
MOSI
CS
GND

MagoKimbra, 2017-12-11:
No the nextion must connect to serial of CPU.
The pin is D14 e D15, D16 e D17, D18 e D19...
Look http://www.adrirobot.it/arduino/arduino_due/immagini/Due-pinout-A4_grande.png in bottom right this image...

jbaiao, 2017-12-11:
Yes, we had already seen that to connect the nextion would have to use the D16 and D17.
I'm asking is if someone knows where to connect the external module of the sd card

MagoKimbra, 2017-12-11:
Ops sorry...

prasarnh, 2017-12-26:
hello sir, can i use with enhance nextion 7" instead of 4.3",thanks

prasarnh, 2017-12-28:
i just found hmi file for 7" and recompile with nextion studio to use with 7" enhance hope this work on smart ramp.

jbaiao, 2018-01-17:
Can anyone tell me which pins correspond to the external micro-sd card connections in smart ramps?
5V
SCK
MISO
MOSI
CS
GND

iosonopersia, 2018-08-25:
@jbaiao please refer to [this image](https://reprap.org/mediawiki/images/8/8d/SMART_RAMPS_1.42.Sch.png). I guess your external SDCard module is connected to Aux3 /Aux2 connector... so find in the image the pins that are connected to the module and extrapolate their pin names.

closed 2018-08-25 by iosonopersia