MKFirmware/MK4duo #339: Contrast for dogm displays

maukcc, 2017-12-08:
### Firmware version
4.3.29_dev

### My setup
arduino due with normal dogm support

The display works but the contrast is not being set from the firmware. 
If I change the contrast in the U8glib.h in:

> uint8_t setContrast(uint8_t contrast) { cbegin(); return u8g_SetContrast(&u8g, contrast); }

to

> uint8_t setContrast(uint8_t contrast) { cbegin(); return u8g_SetContrast(&u8g, 90); }

it works

I am not a coder but I think the U8glib wants a  uint8_t but gets  uint16_t from the firmware and does not know what to do with it.

Also, when you choose one of the preset graphical displays, you get a lot of errors from the ultralcd.cpp that have to do with contrast and  uint16_t

iosonopersia, 2017-12-08:
Hi @maukcc ! Thank you so much for your feedback! @MagoKimbra should have already fixed this error (half an hour ago). Please download again from master branch and retry. Let us know if it works. Bye!

maukcc, 2017-12-08:
Works perfect now with CARTESIO_UI !!

The new files have

> #include "../../../mk4duo.h"

needs to be 

> #include "../../../MK4duo.h"

iosonopersia, 2017-12-08:
Also the "mk4duo/MK4duo" issue was fixed by @MagoKimbra . I think that now this issue can be closed, either by @maukcc or by @MagoKimbra

closed 2017-12-08 by maukcc