MKFirmware/MK4duo #343: scara mechanics support is broken

byebyedizzy, 2017-12-10:
Version from Configuration_version.h:
#define SHORT_BUILD_VERSION "4.3.3"

Compilation errors:

In file included from C:\Temp\arduino_build_714754\sketch\src\core\commands\gcode/gcode.h:42:0,

                 from C:\Temp\arduino_build_714754\sketch\src\core\commands\commands.cpp:30:

C:\Temp\arduino_build_714754\sketch\src\core\commands\gcode/calibrate/g28.h: In function 'void gcode_G28()':

C:\Temp\arduino_build_714754\sketch\src\core\commands\gcode/calibrate/g28.h:31:47: error: 'class Scara_Mechanics' has no member named 'Home'

 inline void gcode_G28(void) { (void)mechanics.Home(false); }

                                               ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:28:0,

                 from C:\Temp\arduino_build_714754\sketch\src\core\commands\../../../MK4duo.h:37,

                 from C:\Temp\arduino_build_714754\sketch\src\core\commands\commands.cpp:29:

C:\Temp\arduino_build_714754\sketch\src\core\commands\gcode/host/m115.h: In function 'void gcode_M115()':

C:\Temp\arduino_build_714754\sketch\src\core\commands\../../../src/lcd/language/language.h:101:174: error: expected ')' before 'MACHINE_TYPE'

 #define MSG_M115_REPORT                     "FIRMWARE_NAME:MK4duo " SHORT_BUILD_VERSION " FIRMWARE_URL:" FIRMWARE_URL " PROTOCOL_VERSION:" PROTOCOL_VERSION " MACHINE_TYPE:" MACHINE_TYPE " EXTRUDER_COUNT:" STRINGIFY(EXTRUDERS) " UUID:" MACHINE_UUID "\n"

                                                                                                                                                                              ^

C:\Temp\arduino_build_714754\sketch\src\core\commands\gcode/host/m115.h:36:3: note: in expansion of macro 'SERIAL_EM'

   SERIAL_EM(MSG_M115_REPORT);

   ^

C:\Temp\arduino_build_714754\sketch\src\core\commands\gcode/host/m115.h:36:13: note: in expansion of macro 'MSG_M115_REPORT'

   SERIAL_EM(MSG_M115_REPORT);

not96, 2018-01-21:
I have the same problem.. I can't compile

Power3DPrinting, 2018-04-08:
Likewise, In file included from sketch\src/core/mechanics/mechanics.h:382:0,

                 from sketch\MK4duo.h:120,

                 from C:\Users\calld_000\Downloads\MK4duo-master\MK4duo-master\MK4duo\MK4duo.ino:301:

sketch\src/core/mechanics/scara/scara_mechanics.h:97:12: error: 'void Scara_Mechanics::do_blocking_move_to(const float&, const float&, const float&, const float&)' marked override, but does not override

       void do_blocking_move_to(const float &lx, const float &ly, const float &lz, const float &fr_mm_s/*=0.0*/) override;

            ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

iosonopersia, 2018-08-06:
Hi @byebyedizzy , @not96  and @Power3DPrinting . We are perfectly aware of this. The problem is that we're missing the code of some important functions of the SCARA mechanics. Quite some time ago I tried to get in a better support for this type of mechanics but it's a really long and hard work, so something is still missing.

Actually, the only thing I can say to you at the moment is **to use the newest version of the original Marlin**, which should support very well SCARA printers. I'm so sorry. We hope to get full SCARA support in the future.

closed 2018-08-06 by iosonopersia

ricardocjcosta, 2019-01-26:
I would also like to try to implement this firmware on a scara using Arduino Mega2560 / Ramps1.4.
It is still under construction. However, when trying to compile, I get the following list of errors. I do not have many programming skills. I tried to figure out how I could solve it but I could not.
I saw that Power3DPrinting was able to do the compilation and load it to the edge, even though it got errors similar to these.
Is it possible to have an explanation of how you did it?
Thank you

sketch\src\core\mechanics\scara_mechanics.cpp: In static member function 'static void Scara_Mechanics::factory_parameters()':

sketch\src\core\mechanics\scara_mechanics.cpp:76:32: error: assignment of read-only location 'Mechanics::homing_feedrate_mm_s[((int)i)]'

       homing_feedrate_mm_s[i]  = pgm_read_float(&tmp_homefeedrate[i]);

                                ^

sketch\src\core\mechanics\scara_mechanics.cpp: In static member function 'static void Scara_Mechanics::do_homing_move(AxisEnum, float, float)':

sketch\src\core\mechanics\scara_mechanics.cpp:532:36: error: 'struct mechanics_data_t' has no member named 'home_dir'

     const bool is_home_dir = (data.home_dir[axis] > 0) == (distance > 0);

                                    ^

sketch\src\core\mechanics\scara_mechanics.cpp:548:38: error: 'sync_plan_position_mech_specific' was not declared in this scope

     sync_plan_position_mech_specific();

                                      ^

sketch\src\core\mechanics\scara_mechanics.cpp: In static member function 'static void Scara_Mechanics::homeaxis(AxisEnum)':

sketch\src\core\mechanics\scara_mechanics.cpp:1065:67: error: 'struct mechanics_data_t' has no member named 'home_dir'

     mechanics.do_homing_move(axis, 1.5f * max_length[axis] * data.home_dir[axis]);

                                                                   ^

sketch\src\core\mechanics\scara_mechanics.cpp:1068:29: error: 'struct mechanics_data_t' has no member named 'home_dir'

     const float bump = data.home_dir[axis] * (

                             ^

sketch\src\core\mechanics\scara_mechanics.cpp:1091:38: error: 'sync_plan_position_mech_specific' was not declared in this scope

     sync_plan_position_mech_specific();

                                      ^

exit status 1
In file included from sketch\src\core\commands\gcode/gcode.h:48:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m92.h: In function 'void gcode_M92()':

sketch\src\core\commands\gcode/config/m92.h:51:17: error: 'class Scara_Mechanics' has no member named 'print_M92'

       mechanics.print_M92();

                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:51:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m201.h: In function 'void gcode_M201()':

sketch\src\core\commands\gcode/config/m201.h:43:17: error: 'class Scara_Mechanics' has no member named 'print_M201'

       mechanics.print_M201();

                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:52:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m203.h: In function 'void gcode_M203()':

sketch\src\core\commands\gcode/config/m203.h:43:17: error: 'class Scara_Mechanics' has no member named 'print_M203'

       mechanics.print_M203();

                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:53:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m204.h: In function 'void gcode_M204()':

sketch\src\core\commands\gcode/config/m204.h:47:17: error: 'class Scara_Mechanics' has no member named 'print_M204'

       mechanics.print_M204();

                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:54:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m205.h: In function 'void gcode_M205()':

sketch\src\core\commands\gcode/config/m205.h:50:17: error: 'class Scara_Mechanics' has no member named 'print_M205'

       mechanics.print_M205();

                 ^

ricardocjcosta, 2019-01-26:
Fixed on today's update! :)