MKFirmware/MK4duo #345: Compiling error

cinealfa, 2017-12-11:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

Cant compile with without SD eeprom enabled

### Compile errors
Archiving built core (caching) in: /var/folders/y1/gqdtkcqj1cn3lp7d480c8z6m0000gp/T/arduino_cache_59003/core/core_arduino_sam_arduino_due_x_aab30a8a57d380c6e82901ec34381a2c.a
sketch/src/core/eeprom/eeprom.cpp.o: In function `EEPROM::read_data(int&, unsigned char*, unsigned short, unsigned short*)':
sketch/src/core/eeprom/eeprom.cpp:332: undefined reference to `eeprom_read_byte(unsigned char*)'
sketch/src/core/eeprom/eeprom.cpp.o: In function `EEPROM::write_data(int&, unsigned char const*, unsigned short, unsigned short*)':
sketch/src/core/eeprom/eeprom.cpp:311: undefined reference to `eeprom_read_byte(unsigned char*)'
sketch/src/core/eeprom/eeprom.cpp:312: undefined reference to `eeprom_write_byte(unsigned char*, unsigned char)'
sketch/src/core/eeprom/eeprom.cpp:313: undefined reference to `eeprom_read_byte(unsigned char*)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Due (Native USB Port).
### Firmware version
MK4duo's versions that have this issue. 4.3.3

### My setup
Ramps FD lcd 2004 with I2c eeprom
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

[Configuration overall.pdf](https://github.com/MKFirmware/MK4duo/files/1549647/Configuration.overall.pdf)

MagoKimbra, 2017-12-11:
Ramps Fd haven't EEPROM I2C, this is your modification.. You must insert @define I2C_EEPROM in your configuration..

cinealfa, 2017-12-11:
ah ok thanks!!

closed 2017-12-11 by cinealfa