MKFirmware/MK4duo #346: Thermocouple MAX31855 wrong readings

pbrgle1, 2017-12-12:
Hi. 
I'm getting wrong readings with my K type thermocouple connected to Ultratronics 1.0
It's reading about 72 deg when it's really 18 deg in the room.
The other day it was reading 32 deg when there were 5 deg in the room.

I'm using the 4.3.3 version from the configurator

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1552453/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1552454/Configuration_Pins.txt)

pbrgle1, 2017-12-12:
I noticed that the temperature was 4 times higher than it should be. After some research i found that the MAX31855 has a resolution of 0.25 deg. So that got me thinking that there has to be something in the firmware where a value gets multiplied by 0.25 or something

I have zero experience with programming but i looked at the sensor.cpp file and compared it to the test firmware from reprapworld.com.

**In the test firmware i noticed the following:**

    if (value & 0x7) {
      return NAN;
    }
    else if(value & 0x80000000) {
      value = 0xFFFFC000 | ((value >> 18) & 0x00003FFFF);
    }
    else {
      value >>= 18;
    }
  
    return value* 0.25;
  }

**The value* 0.25 got my atention so i looked at the sensor.cpp file if there is something similar there:**

// Process temp
    if (data & 0x00010000)
      return 20000; // Some form of error.
    else {
      data = data >> MAX31855_DISCARD_BITS;
      temperature = data & 0x00001FFF;

I tried changing it to  

temperature = 0.25 * (data & 0x00001FFF);

And now the temperatures seem to be correct, but the resolution is still at 1 deg...
Any ideas?

[image]

pbrgle1, 2017-12-12:
Scratch that... Found something else:

  #if ENABLED(SUPPORT_MAX31855)
    if (s_type == -3)
      return read_max31855(this->pin);
  #endif
  #if ENABLED(SUPPORT_MAX6675)
    if (s_type == -2)
      return 0.25 * read_max6675(this->pin, h);

**So i changed it to:**

  #if ENABLED(SUPPORT_MAX31855)
    if (s_type == -3)
      return 0.25 * read_max31855(this->pin);
  #endif
  #if ENABLED(SUPPORT_MAX6675)
    if (s_type == -2)
      return 0.25 * read_max6675(this->pin, h);

Now the resolution is 0.25 deg :)

MagoKimbra, 2017-12-12:
Thank's!!

commit 1d5ced86 (2017-12-12): Minor fix v4.3.3 and fix #346
closed 2017-12-12 by MagoKimbra