MKFirmware/MK4duo #347: Support RepRapDiscount Full Graphic Smart Controller is  broken

ozzysv, 2017-12-12:
Im installed version 4.3.3 and found that the display does not work. On the screen, rubbish
Equipment:  Ardunio DUE + RURAMSMPS4DUE  
On the latest version of the developer everything worked.
Forced to be rolled back to the saved version  4.3.29 dev

wesk-ms, 2017-12-12:
Same problem here.  Random garbage on the 128x64 lcd displays.

DUE+Radds..  4.3.3 built from online configurator.

MagoKimbra, 2017-12-12:
ok checked it!

wesk-ms, 2017-12-18:
Just FYI...I downloaded a zip repository earlier today and the issue still exists.

dembach, 2017-12-18:
on the display , it looks like a heart, caros and some chinese characters , right ?

had the same issues..... just wait ,it seems to be a routine to save to flash.... you programmed via programming port ?

wesk-ms, 2017-12-19:
Yes, lots of random garbage  changing on the display once a second.  Some are actual characters...most of it is just random pixels.  Since I normally use Octoprint with the DUE and RADDS, I've tried flashing through the programming port and the OTG port, as well.  Makes no difference, 4.3.3 no longer works with the 128x64 LCD for me (3-4 different builds/downloads now).  4.3.29 works fine.

4.3.29 has some strange-ness with two Z steppers not being in 100% sync (checked w/O-scope) and on Core XY, I get slight Y movement when only running X axis...and vice versa.  4.3.3 doesn't have either of these problems (yes, same stepper settings).  So, just checking to see if there was an update on the LCD issue with 4.3.3.

I just didn't understand what @MagoKimbra meant by "checked it".  Did he reproduce the issue?  Did he "check" in new code to fix the issue?  :)  That's why I bumped this thread a day ago.

@ozzysv -- are you still seeing the LCD issue on 4.3.3??

Thanks!

MagoKimbra, 2017-12-19:
I do not have the display to be mounted for a card with the DUE, I have normal cards to try. So control means verifying the differences between the old fw and the new ...

MagoKimbra, 2017-12-19:
Try now please!!!

ozzysv, 2017-12-19:
Display has earned

Hans007a, 2017-12-19:
The display don't starts with RAMPS-VD v1with the last fix

MagoKimbra, 2017-12-19:
:( OMG!!!

wesk-ms, 2017-12-19:
Hi,

Still not working correctly on DUE+RADDS after your latest change to ultralcd_st7920_u8glib_rrd_AVR.cpp.

I will say that the MK logo and info screen is visible for a few seconds, but there's a large number of random pixels turned on too.  This causes the info screen to be barely legible. And within a few seconds the LCD starts refreshing with garbage characters and in just a few seconds in total, we're back to just garbage on the LCD.

Also, the display update rate seems faster...maybe once every .5 sec, instead of once per second.  Not 100% certain about that.

My opinion is that you're close and I think it's a timing issue.  I didn't spend much time looking at your change, but it seemed to be somewhat timing related.

FYI, I've also tried adjusting the display timeout setting in the configuration files.  From disabled, to 20us, up to 250us.  No change.  I don't have the code in front of me right this minute but I think the setting was the dog_m SPI timeout (or something like that).  :)

Would a short video of what the display is doing help at all?

Thanks again!!
Wes

MagoKimbra, 2017-12-20:
Try now!!!

Hans007a, 2017-12-20:
I can't copile because this error:

```
sketch\src\core\printer\printer.cpp: In static member function 'static void Printer::IDLE_OOZING_retract(bool)':

sketch\src\core\printer\printer.cpp:883:72: error: 'class Tools' has no member named 'volumetric_multiplier'

       mechanics.current_position[E_AXIS] += IDLE_OOZING_LENGTH / tools.volumetric_multiplier[tools.active_extruder];

                                                                        ^

sketch\src\core\printer\printer.cpp:894:101: error: 'class Tools' has no member named 'volumetric_multiplier'

       mechanics.current_position[E_AXIS] -= (IDLE_OOZING_LENGTH+IDLE_OOZING_RECOVER_LENGTH) / tools.volumetric_multiplier[tools.active_extruder];
```
this //#define VOLUMETRIC_EXTRUSION
is off...

wesk-ms, 2017-12-21:
Still not working with latest commits.

Hans007a, 2017-12-21:
Thank You for the volumetric fix, now I can compile the source, but the display still don"t works :(

ozzysv, 2017-12-23:
I checked again in more detail. 
Yes, in version 4.3.3 there is a problem with timings. 
The frequency of SPI  is 762 kHz.
[https://s26.postimg.org/d6752qua1/image.png](url)
 While in the version 4.3.29 dev  the frequency is 160 kHz
[https://s26.postimg.org/k9f0id7fd/3_29.png](url)

But, the display works. 
[https://s26.postimg.org/wo1sip6nd/IMG_20171223_215807.jpg](url)
The result most likely depends on the quality of the hardware.

The display is very fast. The initial screen saver is shown only 0.3 seconds
The parameter  DOGM_SPI_DELAY_US 5 has no effect on the situation ((

**I think that you need to check the pause code**

Hans007a, 2017-12-24:
It looks that my display is not a good quality, because it don't starts :(

Hans007a, 2017-12-27:
display works again THANKS! :-)

I fund another two bugs, but I will open new issues

and Merry Christmas and Happy New Year!

Hans

wesk-ms, 2017-12-27:
FYI...128x64 LCD still not working on 4,3,3.  DUE+RADDS. I'm happy to provide any additional configuration files and pictures of the display, itself.

ozzysv, 2018-01-02:
Today I downloaded the firmware from the online configurator and

[https://s26.postimg.org/5mybd1aop/IMG_20180102_183823.jpg](url)

**this full  shit**

iosonopersia, 2018-01-02:
A fix is being proposed now in the Marlin repo ([link](https://github.com/MarlinFirmware/Marlin/pull/9009)). This could be easily implemented also in MK4duo, if it does turn out to solve the issue

jf1452, 2018-01-07:
I'm just building a new printer, and thought that the display issue was my bad work; seems I am not alone. I can get it working by pressing hardware reset a couple of times, then things work normally. Looking forward to a software fix.

jmlcatarino, 2018-01-11:
Has anyone managed to get one work with a Smart Ramps board and DUE?

wesk-ms, 2018-01-11:
Not in 4.3.3 or in 4.3.29 in builds created in the past 3 weeks or so.  Prior to that it was rock solid.

I decided to just go with a Nextion display, unfortunately I chose the 3.2 inch enhanced display, and just my luck there isn't any firmware available for the 3.2”.  The 2.8”, 4.3”, and 7” all have firmware, but not the 3.2”

Started porting the larger size to the smaller display…OMG what a pain that is when you don't have all the bitmaps.

Anyone have an easy way to build the 3.2” Nextion firmware that doesn't require starting from scratch?

jf1452, 2018-01-15:
Just tested this yesterday, and the problem now seems to be resolved. Logo display remains on screen for a few seconds, and there is no longer any display corruption. Thanks for fixing this problem.

I'm using DUE with RAMPS-FD modified to v2 and RepRapDiscount Full Graphic Smart Controller

MagoKimbra, 2018-01-15:
Ok thank's

jmlcatarino, 2018-01-15:
Ok guys, just found out how to put the RepRapDiscount Full Graphic Smart Controller working with the Smart Ramps board. The problem is not a FW problem, is a HW problem. Meaning, you need to get yourself an adaptor like the one used on Ramps 1.4 but with modifications to suite the Smart Ramps board, because they're not direct fit. 
So for me, no issues here. :)

wesk-ms, 2018-01-15:
It isn't a hardware problem when it worked fine for months and then stopped working for people when they moved to a certain build of MK or Marlin.  Known issue in the Marlin repo.  Also, not everyone uses the pre-made display boards, so while I'm glad to hear it was a connector issue for you, that isn't the fix for everyone.  I'm using a plain lcd soldered directly to a custom wiring harness.  that has the same SSD7xxx lcd controller chip like all the reprap discount display boards.

Personally, I think the timing issue has mostly been resolved, I still see it get out of sync on occasion.  Lots better, for sure.

jmlcatarino, 2018-01-16:
@wesk-ms I was not referring to timing issues on LCD, I was referring that as far as my "issue", which was only about connecting and getting the LCD working on a Smart Ramps board +Due, I managed to find what was the no-problem. In fact it is just lack of general support for that particular board. The guys did a good job developing the board, got a screen working but forgot that most of us uses a Ramps 1.4 and the cheap RepRapDiscount Full Graphic LCD Smart Controller. There are only a few pics where we can see a screen connected but without a connector like the older versions. So I tried to connect it like in the picture and changing the pin assignment, but with no avail. I'll have to hack a "regular" LCD adaptor in order to make it work with this one, because it's not a fit match. 
Other than that, I had an FD and this same LCD screen and never noticed any issues.

dell9, 2018-01-22:
Had the same problem with the v4.3.3 firmware. Was able to use LCD after shielding of connector wires. I took aluminium foil, put wires into it, and covered shielded wires with Heat shrinkage tube

ozzysv, 2018-02-01:
Hello
on version 4.3.4  the display works well
Thank you

closed 2018-02-01 by ozzysv

Lippo89, 2018-10-01:
Hi, is there somewhere istructions how to modify the ramps 1.4 LCD adaptor to the smart ramps? Do I need also modify the board file?

jmlcatarino, 2018-10-01:
I did change it but didn't write anything about it. The only thing I managed to get was a EagleCad design to make them. There's a difference between the regular adaptor and the smart ramps adaptor that is only about distance of the pins. And I think a minor change to the pins also, power related. 
I can send you them files if you want to take a go on them.

Lippo89, 2018-10-01:
Hi, I cut one of my ramps 1.4 LCD adaptor to two pieces and then I soldered wires between them. But I mange to get only backlight and beeper working. So if you can send files to me I rally appreciate it. Can you send link where I can grab files or how you share files?

jmlcatarino, 2018-10-01:
Hi there, sure but I don't seem to be able to post it here and I'm not sue how to do it. 
SmartRamps LCD Adaptor.brd
SmartRamps LCD Adaptor.sch

Lippo89, 2018-10-01:
Thanks for adaptor files, but I don't have any software for reading those. Do you have any picture form the adaptor scheme? Can you pelase send your smart ramps board.h file.

jmlcatarino, 2018-10-01:
You'd had to have Eagle Cad installed. I'll just create a pdf and post it here, no that I know how to do it. :)

Lippo89, 2018-10-01:
Big thanks to you! :) It looks like I'm missing few wires. I try to do modifications acc. your scheme.

jmlcatarino, 2018-10-01:
It works because I did it with a perf board and some ribbon cable wires and is still there. :)

Lippo89, 2018-10-02:
Now I got my screen working. Thanks for the help @jmlcatarino ! :)

jmlcatarino, 2018-10-02:
Glad to know that it worked for you. :)

MagoKimbra, 2018-10-02:
Thank's @jmlcatarino !!!

jmlcatarino, 2018-10-02:
Thank you  @MagoKimbra   for your good work on this firmware. ;)