MKFirmware/MK4duo #349: MAXTEMP Error

pbrgle1, 2017-12-13:
I noticed a strange and potentially dangerous bug.

I just got my thermocouple readings working, but now i'm getting MAXTEMP errors even thought the temperature is under the max threshold (320C in my case).

The error appears when i execute the extrude command and sometimes it's even spontaneous when the heaters are on. The biggest problem is that sometimes the heaters DON'T turn OFF after the error and the temperature keeps rising.

[image]

pbrgle1, 2017-12-15:
I want to point out that the issue with the rising temperature is not resolved.

READ: ok T:213.00 /230 B:10.50 /0 B@:0 @:255 T0:17.00 /0 @0:0 T1:213.00 /230 @1:255 MCU min:35.29, current:40.99, max:40.99
READ: wait
SENT: M105
READ: ok T:214.75 /230 B:10.55 /0 B@:0 @:255 T0:17.00 /0 @0:0 T1:214.75 /230 @1:255 MCU min:35.29, current:40.77, max:41.15
READ: wait
SENT: M105
READ: ok T:216.25 /230 B:10.50 /0 B@:0 @:255 T0:17.00 /0 @0:0 T1:216.25 /230 @1:255 MCU min:35.29, current:40.84, max:41.15
READ: wait
SENT: M105
READ: wait
READ: ok T:218.00 /230 B:10.55 /0 B@:0 @:255 T0:17.00 /0 @0:0 T1:218.00 /230 @1:255 MCU min:35.29, current:40.99, max:41.15
READ: wait
SENT: M105
READ: wait
READ: ok T:219.75 /230 B:10.61 /0 B@:0 @:255 T0:17.00 /0 @0:0 T1:219.75 /230 @1:255 MCU min:35.29, current:40.99, max:41.30
READ: wait
SENT: G91
SENT: G1 E1.0000 F240
SENT: G90
READ: ok
READ: ok
READ: ok
SENT: G91
SENT: G1 E1.0000 F240
SENT: G90
READ: ok
READ: ok
READ: ok
SENT: M105
READ: ok T:220.75 /230 B:10.61 /0 B@:0 @:0 T0:16.50 /0 @0:0 T1:220.75 /230 @1:0 MCU min:35.29, current:41.37, max:41.45
READ: Error:MAXTEMP triggered, system stopped! Heater_ID: 1
READ: Error:Printer halted. kill() called!
SENT: M105
SENT: M105

After this, the temperature keeps rising and the pin only shuts off when i reconnect.
Very dangerous!

MagoKimbra, 2017-12-15:
When it detects the temperature error, the first thing it does is turn off all the heaters then it goes into kill mode, which you can select either type 1 or type 2. Type 1 writes the kill, but it does not do anything, the type 2 reset the card. It is impossible for the heaters to remain on.
The error occurs as soon as the current temperature is greater than the max temp. The temperature value that I read 220 is vermanete the current one?

pbrgle1, 2017-12-15:
Well it's very possible because it almost burned my hotend. There was smoke comming out.

MagoKimbra, 2017-12-15:
I have no doubt that this has happened, the board is very probably blocked, if you lock the little processor you can do, the only thing is to activate the watch dog ...

pbrgle1, 2017-12-15:
How do i set type 2 for the kill mode?

MagoKimbra, 2017-12-15:
Try it, but the problem is with SAM3xe the haven't reset software...

pbrgle1, 2017-12-15:
it doesn't work...

What can i do to disable the heaters after an error?

cjsoong, 2017-12-15:
SORRY My English is not good...
我想應該是你的PC online contorl panel 的問題.
因為DUE main board reset 但是UART ->USB  還是正常工作,所以當mb reset 完後 UART 會繼續傳送加熱的Gcode cmd 給MB ,所以 加熱還是繼續..
我曾在meg2560+ramps1.4 遇到類似問題..按下KILL鍵,但RESET結束後 , PC online control  panel 繼續送列印CMD , XYZ 馬達繼續移動.

****Google Translate****
I think it should be your PC online contorl panel problem.
Because DUE main board reset but UART -> USB is still working, so when mb reset finished UART will continue to send heated Gcode cmd to MB, so the heating continues.
I encountered a similar problem in meg2560 + ramps1.4 .. Press the KILL key, but after the RESET is finished, the PC online control panel continues to send the CMD, and the XYZ motor continues to move.

pbrgle1, 2017-12-17:
Hi.

Right now everything seems to be working fine, except the heaters dont get disabled after an error.

This is a major issue because i'm afraid to ever leave the room when i'm printing.

Is there something that can be done so that the heaters get disabled after a error?

EPES-Rades, 2018-02-05:
I have same issue. If I disconnect thermistor Pronterface write Error but heaters still heating.

My configuration:
arduino due, ramps-fd (programing port). firmware V4.3.4.

I tried to change below parametrs but without result:

#define INVERTED_HEATER_PINS false
#define INVERTED_BED_PIN false

#define THERMAL_PROTECTION_HOTENDS true
#define THERMAL_PROTECTION_BED true

#define KILL_METHOD 0

When I tested same configuration with repetier firmware thermal protection worked well. But I have the best guality of 3D printing with MK4Duo :-)

commit c1c684d0 (2018-02-05): Fix #349
closed 2018-02-05 by MagoKimbra