MKFirmware/MK4duo #350: Cannot Compile with TMC2130

dembach, 2017-12-17:
### Issue description
cannot compile with TMC2130 activated.

### Compile errors
`Arduino: 1.8.5 (Mac OS X), Board: "Arduino Due (Programming Port)"

sketch/src/core/printer/printer.cpp: In static member function 'static void Printer::manage_inactivity(bool)':
sketch/src/core/printer/printer.cpp:787:27: error: 'tmc2130_checkOverTemp' was not declared in this scope
     tmc2130_checkOverTemp();

                           ^
exit status 1
`
### Firmware version
4.3.3 of 17.12.2017 

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
changed configuration_overall : 
#define HAVE_TMC2130

changed the CS Pins in 1403.h (RAMPS_FD_v1):
#define ORIG_X_CS_PIN              54
#define ORIG_Y_CS_PIN              55
#define ORIG_Z_CS_PIN              56
#define ORIG_E0_CS_PIN             57

changes in Configuration_Motor_Driver.h:
#if ENABLED(HAVE_TMC2130)

  // CHOOSE YOUR MOTORS HERE, THIS IS MANDATORY
  #define X_IS_TMC2130
  #define Y_IS_TMC2130
  #define Z_IS_TMC2130
  #define E0_IS_TMC2130

markv92, 2017-12-18:
You have to use: Arduino Due (Native USB Port) to compile and flash this firmware on DUE...
Try ;) i hope it works

iosonopersia, 2017-12-18:
@MagoKimbra The problem is that (in MK4duo.h) **feature/tmc2130/tmc2130.h** is included after **core/printer/printer.h**, which needs it.

dembach, 2017-12-18:
@markv92 it does not matter, for wich port you compile....it is just the purpose of the upload how the image is prepared....

dembach, 2017-12-18:
@iosonopersia i moved the 

`#if

 ENABLED(HAVE_TMC2130)
  #include <TMC2130Stepper.h>
#endif`

up to line 106, just after the HAL Modules...
but it does not work also.. same message....

markv92, 2017-12-18:
@dembach ok, i told you that because i got some errors tring to compile with programming port the initial 4.3.3 release and those errors disappears choosing native usb port.
With current 4.3.3 bugfix both due ports works fine

iosonopersia, 2017-12-18:
@dembach that's not the line I was talking about... The right line to move is line 142 "#include src/feature/tmc2130/tmc2130.h"

dembach, 2017-12-18:
`#include "src/feature/tmc/tmc.h"`

iosonopersia, 2017-12-18:
Oooops!!! My local copy of the repo was not updated! Silly error by me... Anyway, go to printer.cpp at line 787 and replace `tmc2130_checkOverTemp();` with `monitor_tmc_driver();`. This should make it work. If that's the case, we will fix it in the repo.

dembach, 2017-12-18:
even worse:
`Arduino: 1.8.5 (Mac OS X), Board: "Arduino Due (Native USB Port)"

Archiving built core (caching) in: /var/folders/ll/12w1k0cx67j5ynw4pf5tghhr0000gn/T/arduino_cache_364867/core/core_arduino_sam_arduino_due_x_ffb517edfcbd46ae0a5aabe083e2c4d8.a
sketch/src/core/commands/commands.cpp.o: In function `gcode_M911()':
sketch/src/core/commands/gcode/feature/m911_m915.h:39: undefined reference to `void tmc_report_otpw<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/feature/m911_m915.h:42: undefined reference to `void tmc_report_otpw<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/feature/m911_m915.h:45: undefined reference to `void tmc_report_otpw<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/feature/m911_m915.h:48: undefined reference to `void tmc_report_otpw<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/commands.cpp.o: In function `gcode_M912()':
sketch/src/core/commands/gcode/feature/m911_m915.h:61: undefined reference to `void tmc_clear_otpw<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/feature/m911_m915.h:68: undefined reference to `void tmc_clear_otpw<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/feature/m911_m915.h:72: undefined reference to `void tmc_clear_otpw<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/feature/m911_m915.h:76: undefined reference to `void tmc_clear_otpw<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/commands.cpp.o: In function `gcode_M906()':
sketch/src/core/commands/gcode/config/m906.h:63: undefined reference to `void tmc_set_current<TMC2130Stepper>(TMC2130Stepper&, char const*, int)'
sketch/src/core/commands/gcode/config/m906.h:64: undefined reference to `void tmc_get_current<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/config/m906.h:71: undefined reference to `void tmc_set_current<TMC2130Stepper>(TMC2130Stepper&, char const*, int)'
sketch/src/core/commands/gcode/config/m906.h:72: undefined reference to `void tmc_get_current<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/config/m906.h:79: undefined reference to `void tmc_set_current<TMC2130Stepper>(TMC2130Stepper&, char const*, int)'
sketch/src/core/commands/gcode/config/m906.h:80: undefined reference to `void tmc_get_current<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/commands/gcode/config/m906.h:87: undefined reference to `void tmc_set_current<TMC2130Stepper>(TMC2130Stepper&, char const*, int)'
sketch/src/core/commands/gcode/config/m906.h:88: undefined reference to `void tmc_get_current<TMC2130Stepper>(TMC2130Stepper&, char const*)'
sketch/src/core/printer/printer.cpp.o: In function `Printer::manage_inactivity(bool)':
sketch/src/core/printer/printer.cpp:787: undefined reference to `monitor_tmc_driver()'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Native USB Port).

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

MagoKimbra, 2017-12-18:
I think fixed it! Please try and write me if have problem!

dembach, 2017-12-18:
@MagoKimbra 👍 
Compile was successful.....!!!!!

closed 2017-12-18 by dembach