MKFirmware/MK4duo #352: Arduino due not booting if TMC2130 is activated

dembach, 2017-12-18:
Hi,
got a problem with actual 4.3.3, if in my config TMC2130 is active, the boot of the due (without shield) is not completed. (no monitor output on programming port)
see attached configs, board is RAMPS FD v1 (1403), x/y/z are TMC2130

[1403.txt](https://github.com/MKFirmware/MK4duo/files/1569511/1403.txt)
[Configuration_Motor_Driver.txt](https://github.com/MKFirmware/MK4duo/files/1569514/Configuration_Motor_Driver.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1569516/Configuration_Pins.txt)
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1569517/Configuration_Overall.txt)
thanks
Markus

dembach, 2017-12-19:
@MagoKimbra any idea ?

MagoKimbra, 2017-12-19:
I do not know... The TMC2130 libraries are not made for the DUE processor, maybe they do not handle the SPI well.

dembach, 2017-12-28:
@MagoKimbra , 
Bherit found something:
#268

dembach, 2017-12-29:
Fixed with the modified Libraries of @Beherith , @teemalut is informed via Ticket ...

closed 2017-12-29 by dembach