MKFirmware/MK4duo #353: pins for Extruder Encoder Control

maukcc, 2017-12-22:
What pins are used/defined for Extruder Encoder Control ?
Is it just the FIL_RUNOUT_PIN?
And then that is just hooked to the A or B signal of the encoder? (because we do not care about direction of movement)

MagoKimbra, 2017-12-22:
No the FIL_RUNOUT is only the end of wire switch, the encoder is another thing and if you have it you associate it with the pin you want, there are no pre-associated pins.
The encoder gives the pulses to the movement of the filament these pulses must correspond to the movement of the stepper, the firmware does the calculation with the entered values, if the filament does not move then the firmware goes on pause ...

maukcc, 2017-12-22:
Perfect.
What are the defined pins for this?
EXTRUDER_ENCODER_A_PIN
EXTRUDER_ENCODER_B_PIN
?

maukcc, 2017-12-22:
Also, is there any delay, like it first needs to empty the buffer.
And then it takes 16 Gcodes to go on pause :)
Is it pause or M600? or user defined script?

MagoKimbra, 2017-12-22:
No one pin why 2 pin??
only E0_ENC_PIN or E1_ENC_PIN, E2_ENC_PIN etc

maukcc, 2017-12-22:
With 1 pin you only get pulses and no direction, so I can imagine that retracts give false readings.

MagoKimbra, 2017-12-22:
Yes only movement no direction!!!

maukcc, 2017-12-22:
AAH and it is extruder dependent. so for 4 extruders we will need 4 inputs.
That is valuable real estate :) and it is only needed for mixing extruders.
Would it be possible to use it with only 1 input for all extruders

alfredanil, 2018-01-08:
How about using i2c bus optical encoder? like Prusa MK3 ;-)

maukcc, 2018-01-09:
I would love to have it i2c :) that would save 4 pins on my overbooked due. 

1 more thing I would like to see:
We have 4 filaments on 4 extruders, and now each filament input has to be designated to a specific extruder. (Input 1 is for extruder 1)
When I want red filament which is in input 3 to go on extruder 1, we need to physically pull the filament all the way out of input 3, and insert it into input 1.
Would it be possible ,on the nextion lcd, to have a screen where you commit input X to extruder Y? 
If someone could program that in, it would be worth 1 of our new (untested :) ) dual FilamentEncoders that give 96 pulses per rotation(or 1puls/0.22mm) with DHT22 to measure moisture levels and temp at the filament roll.