MKFirmware/MK4duo #355: z probe servo not work anet a8

ISU000, 2017-12-25:
hi, i test in all input 27,28,29,30 but with m280 don't move servo (3D Touch) the 3D Touch with marlin 3.1.6 in my anet work well but without nextion 7 lcd.

This is my configuration:
[Configuration_Overall.h.zip](https://github.com/MKFirmware/MK4duo/files/1585936/Configuration_Overall.h.zip)

*Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Tell us here what happened: don't miss any valuable detail, be clear and concise, don't use UPPERCASE or abbreviations which can make understanding what you wrote quite difficult.

### Compile errors
If you have problems compiling MK4duo, report here the error messages Arduino IDE gives you.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
MK4duo's versions that have this issue.

### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

dembach, 2017-12-29:
did you define a z-probe pin in the file 100.h ?
```
//###SERVO
#define SERVO0_PIN                 NoPin
#define SERVO1_PIN                 NoPin
#define SERVO2_PIN                 NoPin
#define SERVO3_PIN                 NoPin

```

if you want to connect a Servo you need to define a pin, as far as i know, PIN 27 is free for use on this Board:
#define SERVO0_PIN        27
But it depends on your Display ..... in your configuration you have got no display.. is that correct ?

`https://github.com/ralf-e/ANET-3D-Board-V1.0`

If you are using the ANET Full Graphics Display (like on A6) , Pin 29 is free

ISU000, 2017-12-30:
Yes I define 27, 28, 29 and 30 in 100.h file but don’t move 3D Touch I write direct in configuration overall define servo0_pin 27,28,29, and 30 and never move 3D Touch or normal servo.

I install marlin 3.1.6 in the same Anet Board and 3D Touch work perfect without nextion 7.

Now  I change board to ramps 1.4+ Arduino mega I install mk4duo 4.3.3 and 3D Touch work perfect with nextion 7.

https://github.com/ralf-e/ANET-3D-Board-V1.0

bloutix, 2018-01-07:
You need to make this modification for anet board

https://github.com/MKFirmware/MK4duo/issues/365

kind regards

PR #367 (2018-01-09): Update SERVO code and some fixes
This pull request is supposed to close the following issues:
- Implement suggestion to fix #365 . This should also fix #355 . Thanks to @bloutix for this
- Little error fix #366 (SCK instead of CK). Thanks to @Hans007a for this
- Created CASE_LIGHT sanitycheck to fix #335 .
closed 2018-01-09 by MagoKimbra