MKFirmware/MK4duo #357: assign multiple pins to a define

maukcc, 2017-12-27:
Is it possible to assign multiple hardware pins to 1 define?
for example
#define ORIG_E0_STEP_PIN           57, 58, 59, 60
This to accommodate parallel printing with multiple extruders. 
Right now we just have the four extruders hardware wired to 1 pin, but it would be more elegant to have separate pins. This way we can switch between 4 extruders making 1 object or 4 extruders making 4 copies at the same time.

MagoKimbra, 2017-12-27:
No not possible this!!!

maukcc, 2017-12-29:
OK, thanks. We will make hardware to do this then.

closed 2017-12-29 by maukcc

MagoKimbra, 2017-12-29:
No you must change this line in stepper_indirection.h
#define E0_STEP_WRITE(STATE) WRITE(E0_STEP_PIN,STATE)
in 
#define E0_STEP_WRITE(STATE) WRITE(E0_STEP_PIN,STATE), WRITE(E1_STEP_PIN,STATE); WRITE(E2_STEP_PIN,STATE); WRITE(E3_STEP_PIN,STATE)

And
#define E0_STEP_PIN 57
#define E1_STEP_PIN 58
#define E2_STEP_PIN 59
#define E3_STEP_PIN 60

maukcc, 2017-12-29:
Thanks, that seems straight forward. We will give it a try, and if it is successful we will make a PR with it.
Something along the lines of:
If defined parallel_printing then...

MagoKimbra, 2017-12-29:
Yes, but the problem is dir and enabled..
Even these pins should be put in parallel

maukcc, 2017-12-29:
Would that not be the same as the step?

MagoKimbra, 2017-12-29:
Yes...

maukcc, 2017-12-29:
I will let you know how it goes next year :)

MagoKimbra, 2017-12-29:
Ok