MKFirmware/MK4duo #358: compile error: u8g nt found

Hans007a, 2017-12-29:
I got a compile error with the new version from 29.12.2017:

```
In file included from sketch\src\lcd\ultralcd.cpp:60:0:

sketch\src\lcd\ultralcd_impl_DOGM.h:170:5: error: 'U8GLIB_ST7920_128X64_CUSTOM_SW_SPI' does not name a type

     U8GLIB_ST7920_128X64_CUSTOM_SW_SPI u8g;

     ^

In file included from sketch\src\lcd\ultralcd_impl_DOGM.h:229:0,

                 from sketch\src\lcd\ultralcd.cpp:60:

sketch\src\lcd\utf_mapper.h: In function 'char charset_mapper(char)':

sketch\src\lcd\utf_mapper.h:29:32: error: 'u8g' was not declared in this scope

   #define HARDWARE_CHAR_OUT(C) u8g.print((char)(C))

                                ^

sketch\src\lcd\utf_mapper.h:665:5: note: in expansion of macro 'HARDWARE_CHAR_OUT'

     HARDWARE_CHAR_OUT(c);

     ^

In file included from sketch\src\lcd\ultralcd.cpp:60:0:

sketch\src\lcd\ultralcd_impl_DOGM.h: At global scope:

sketch\src\lcd\ultralcd_impl_DOGM.h:235:35: error: 'u8g' was not declared in this scope

 u8g_page_t &page = ((u8g_pb_t *)((u8g.getU8g())->dev->dev_mem))->p;

                                   ^

sketch\src\lcd\ultralcd_impl_DOGM.h: In function 'void lcd_setFont(char)':

sketch\src\lcd\ultralcd_impl_DOGM.h:243:29: error: 'u8g' was not declared in this scope

     case FONT_STATUSMENU : {u8g.setFont(FONT_STATUSMENU_NAME); currentfont = FONT_STATUSMENU;}; break;
...
...
```

My board ist DUE with RAMPS-FD v1 with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Regards

Hans

iosonopersia, 2017-12-29:
The error comes from this snippet of code:
``` cpp
#elif ENABLED(U8GLIB_ST7920) && defined(__arm__)
  // RepRap Discount Full Graphics Smart Controller on an ARM target
    U8GLIB_ST7920_128X64_CUSTOM_SW_SPI u8g;

#elif ENABLED(U8GLIB_ST7920)
[continues...]
```

It seems that the `__arm__` macro is defined by the compiler, while it shouldn't (RAMPS-FD is Due-based, it's not an ARM processor). In my opinion, since MK4duo doesn't support ARM cpus, that elif statement could be safely erased from the code @MagoKimbra

MagoKimbra, 2017-12-29:
Delete this line....

Hans007a, 2017-12-30:
Now work's, thank You and Happy New Year!

closed 2017-12-30 by Hans007a