MKFirmware/MK4duo #359: M43 Pin Output incomplete

dembach, 2017-12-29:
```
PIN:  0        RXD                                protected 
PIN:  1        TXD                                protected 
PIN:  2        FAN1_PIN                           Output = 0
PIN:  3        E0_CS_PIN                          Output = 1
PIN:  4        SS_PIN                             Output = 1
PIN:  5        <unused>                           Output = 1
PIN:  6        <unused>                           Output = 1
PIN:  7        <unused>                           Output = 1
PIN:  8        HEATER_BED_PIN                     Output = 1
PIN:  9        HEATER_0_PIN                       Output = 1
PIN: 10        HEATER_1_PIN                       Input  = 1
PIN: 11        HEATER_2_PIN                       Input  = 1
PIN: 12        FAN0_PIN                           Output = 0
PIN: 13        LED_PIN                            Input  = 1
PIN: 14        <unused>                           Input  = 1
PIN: 15        <unused>                           Input  = 1
PIN: 16        LCD_PINS_RS                        Output = 0
PIN: 17        LCD_PINS_ENABLE                    Output = 0
PIN: 18        <unused>                           Input  = 1
PIN: 19        <unused>                           Input  = 1
PIN: 20        <unused>                           Input  = 1
PIN: 21        <unused>                           Input  = 1
PIN: 22        <unused>                           Input  = 1
PIN: 23        LCD_PINS_D4                        Output = 1
PIN: 24        <unused>                           Input  = 1
PIN: 25        LCD_PINS_D5                        Input  = 1
PIN: 26        Z_PROBE_PIN                        Input  = 1
PIN: 27        LCD_PINS_D6                        Input  = 1
PIN: 28        E0_DIR_PIN                         protected 
PIN: 29        LCD_PINS_D7                        Input  = 1
PIN: 30        X_MAX_PIN                          Input  = 1
PIN: 31        BTN_EN2                            Input  = 1
PIN: 32        <unused>                           Input  = 1
PIN: 33        BTN_EN1                            Input  = 1
PIN: 34        Z_MAX_PIN                          Input  = 1
PIN: 35        BTN_ENC                            Input  = 1
PIN: 36        E0_STEP_PIN                        protected 
PIN: 37        BEEPER_PIN                         Input  = 0
PIN: 38        Y_MAX_PIN                          Input  = 1
PIN: 39        <unused>                           Input  = 1
PIN: 40        <unused>                           Input  = 1
PIN: 41        <unused>                           Input  = 1
PIN: 42        E0_ENABLE_PIN                      protected 
PIN: 43        <unused>                           Input  = 1
PIN: 44        Z_ENABLE_PIN                       protected 
PIN: 45        <unused>                           Input  = 1
PIN: 46        Y_ENABLE_PIN                       protected 
PIN: 47        <unused>                           Input  = 1
PIN: 48        X_ENABLE_PIN                       protected 
PIN: 49        SD_DETECT_PIN                      Input  = 1
PIN: 50        <unused>                           Input  = 1
PIN: 51        <unused>                           Output = 0
PIN: 52        <unused>                           Output = 0
PIN: 53        MAX6675_SS_PIN                     Input  = 1
PIN: 54 (A 0)  TEMP_BED_PIN                       
PIN: 55 (A 1)  TEMP_0_PIN                         
PIN: 56 (A 2)  TEMP_1_PIN                         
PIN: 57 (A 3)  TEMP_2_PIN                         
PIN: 58 (A 4)  <unused>                           Input  = 1
PIN: 59 (A 5)  <unused>                           Input  = 1
PIN: 60 (A 6)  <unused>                           Input  = 1
PIN: 61 (A 7)  <unused>                           Input  = 1
PIN: 62 (A 8)  X_DIR_PIN                          protected 
PIN: 63 (A 9)  X_STEP_PIN                         protected 
PIN: 64 (A10)  Y_DIR_PIN                          protected 
PIN: 65 (A11)  Y_STEP_PIN                         protected 
PIN: 66        Z_DIR_PIN                          protected 
PIN: 67        Z_STEP_PIN                         protected 
PIN: 68        <unused>                           Input  = 1
PIN: 69        <unused>                           Input  = 1
PIN: 70        <unused>                           Input  = 1
PIN: 71        <unused>                           Input  = 1
PIN: 72        <unused>                           Input  = 1
PIN: 73        <unused>                           Input  = 1
PIN: 74        MISO_PIN                           Input  = 1
PIN: 75        MOSI_PIN                           Input  = 1
PIN: 76        SCK_PIN                            Input  = 1
PIN: 77        <unused>                           Input  = 1
PIN: 78        <unused>                           Input  = 1

```
Missing PINS are ORIG_X_CS_PIN, ORIG_Y_CS_PIN, ORIG_Z_CS_PIN
```
//###X_AXIS
#define ORIG_X_STEP_PIN            63
#define ORIG_X_DIR_PIN             62
#define ORIG_X_ENABLE_PIN          48
#define ORIG_X_CS_PIN              7  

//###Y_AXIS
#define ORIG_Y_STEP_PIN            65
#define ORIG_Y_DIR_PIN             64
#define ORIG_Y_ENABLE_PIN          46
#define ORIG_Y_CS_PIN              6

//###Z_AXIS
#define ORIG_Z_STEP_PIN            67
#define ORIG_Z_DIR_PIN             66
#define ORIG_Z_ENABLE_PIN          44
#define ORIG_Z_CS_PIN              5

//###EXTRUDER_0
#define ORIG_E0_STEP_PIN           36
#define ORIG_E0_DIR_PIN            28
#define ORIG_E0_ENABLE_PIN         42
#define ORIG_E0_CS_PIN             3
#define ORIG_SOL0_PIN              NoPin
```

MagoKimbra, 2017-12-29:
Ok..

dembach, 2017-12-29:
it is just depending, wich function is assigned... missing functions are ORIG_X_CS_PIN, ORIG_Y_CS_PIN, ORIG_Z_CS_PIN ,  but ORIG_E0_CS_PIN is included... funny... ;-)

MagoKimbra, 2017-12-31:
But now there are all CS pins. What you're missing??

dembach, 2017-12-31:
i will give it a try and take a look if there is something else missing.... thanks @MagoKimbra

closed 2018-01-02 by MagoKimbra