MKFirmware/MK4duo #361: Arduino Due and Smart Ramps - Cant get SDCardreader to work in 4.3.3

Skater2k2, 2018-01-02:
Hello,

I running your nice FW with a Nextion Display and a separate SDCard reader.
Everytime if i try to mount my SDCard with M21, i get the error: SD init error

With the same setup on version 4.3.1x dev, a fork from cxandy, sd mounting works fine. 
I tried the same settings and also your online configurator with no luck. I looked at the Pin Mapping, but i cant see any difference...

Maybe you can help me.

[Settings.zip](https://github.com/MKFirmware/MK4duo/files/1598632/Settings.zip)

MagoKimbra, 2018-01-02:
In 4.3.1 he was using SPI software, now he uses the hardware because they told me about the FD ramps, so are you sure you have attached it to the right MOSI MISO and SCK hardware pins?

Skater2k2, 2018-01-03:
Thank you for your fast response. 

I cant say much about the Ramps FD board, but on my Smart Ramps board the hardware SPI Pins are not connectet with the Due. [image]
On [http://reprap.org/wiki/SMART_RAMPS](url) they say:
Note:SMART RAMPS + DUE only support SW SPI, SMART RAMPS + MEGA2560 support HW SPI.

Is there an easy way to switch back to Software SPI?

Wiring is:
Miso to D50
SCK to D52
Mosi to D51
CS to D53

Skater2k2, 2018-01-03:
I try the following wich works fin for me now:

[image]

It looks a bit crude, but i can live with it. Thanks for the tip with Hardware SPI.

MagoKimbra, 2018-01-03:
O_O

The alternative is delete in src/hal/hal_due/spi_pin_due the check for CS pin 53 and return force SPI software...

#if (SDSS == 4) || (SDSS == 10) || (SDSS == 52) || (SDSS == 53) || (SDSS == 59) || (SDSS == 60) || (SDSS == 77)

Delete SDSS = 53

#if (SDSS == 4) || (SDSS == 10) || (SDSS == 52) || (SDSS == 59) || (SDSS == 60) || (SDSS == 77)

jbaiao, 2018-02-23:
I have this board and make work like Skater2k2 did.
But I have a question, how can I use tmc2130 with spi configuration?

MagoKimbra, 2018-02-24:
Use software SPI, in configuration_pins.h you can found this:
```
#if ENABLED(HAVE_TMC2130) && ENABLED(SOFT_SPI_TMC2130)
  #define SOFT_MOSI_PIN 51
  #define SOFT_MISO_PIN 50
  #define SOFT_SCK_PIN  52
#endif
```
You must enabled SOFT_SPI_TMC2130 in Configuration_drivers.h

jbaiao, 2018-02-25:
But this way work, sd card module still work?
Or i have to remove it?

Beherith, 2018-03-02:
Yes, SD card will still work when both TMC2130 and SDcards use software SPI.

jbaiao, 2018-04-26:
Help me please!

After enabled TMC2130, i reflash DUE and dont boot :(
Keep booting and i dont now why!

I use mk4due 4.3.5 and pins:
  #define SOFT_MOSI_PIN 51
  #define SOFT_MISO_PIN 50
  #define SOFT_SCK_PIN  52

I use smart ramps.

iosonopersia, 2018-08-25:
@jbaiao Haven't you already solved your SDCard-TMC2130 SPI connection issue here https://github.com/MKFirmware/MK4duo/issues/466#issuecomment-398061165 ?

closed 2018-08-25 by iosonopersia