MKFirmware/MK4duo #362: Due not bootet if enable tmc2130

Welder26, 2018-01-04:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Tell us here what happened: don't miss any valuable detail, be clear and concise, don't use UPPERCASE or abbreviations which can make understanding what you wrote quite difficult.

### Compile errors
If you have problems compiling MK4duo, report here the error messages Arduino IDE gives you.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
MK4duo's versions that have this issue.

### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

Welder26, 2018-01-04:
Set up is Radds ,Due and Nextion Display 7" Enhanced

Beherith, 2018-01-05:
Use this library instead for MK4DUO tmc2130 support:

https://github.com/Beherith/TMC2130Stepper
Make sure to define the SPI pins you will be using in TMC2130Stepper.cpp

Welder26, 2018-01-05:
I define on Aux 1 the cs pins 

Miso mosi etc on extern sd

Welder26, 2018-01-06:
also not booting ......

Welder26, 2018-01-06:
Is here anyone he has run tmc2130 on Radds ? And can send me the config ?

sh4jonny, 2018-01-21:
@Welder26 i have tmc2130 on hackedRAPMS and DUE
i use it with @Beherith lib. i define pins for Software SPI in lib source - TMC2130Stepper.cpp
SW_SPI TMC_SPI = SW_SPI(4, 6, 5);           - in my case

and in pin file i define CS pins

Welder26, 2018-01-22:
Yes , but who are the spi pin out in RADDS ...

sh4jonny, 2018-01-22:
Use any free pins. Just define it

Welder26, 2018-01-22:
I define any one from aux3 ? And then works?

closed 2018-02-14 by Welder26