MKFirmware/MK4duo #369: Used latest configurator, unfortunately still unable to build..

jf1452, 2018-01-10:
### Issue description
sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_st7920_sw_spi.cpp
line:157:7 add an 'S'! This line will be look:
SCK_pPio->PIO_CODR = SCK_dwMask; //SCK low - needed at power up but not after reset
This line seems correct now. Apologies, but the original compilation issue seems to have returned.

### Compile errors
Arduino: 1.8.5 (Windows 10), Board: "Arduino Due (Programming Port)"

In file included from sketch\src\lcd\ultralcd.cpp:64:0:

sketch\src\lcd\ultralcd_impl_DOGM.h:204:30: error: 'DOGLCD_SCK' was not declared in this scope

     U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

                              ^

sketch\src\lcd\ultralcd_impl_DOGM.h:204:42: error: 'DOGLCD_MOSI' was not declared in this scope

     U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

                                          ^

sketch\src\lcd\ultralcd_impl_DOGM.h:204:55: error: 'DOGLCD_CS' was not declared in this scope

     U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

                                                       ^

sketch\src\lcd\ultralcd_impl_DOGM.h:204:66: error: 'DOGLCD_A0' was not declared in this scope

     U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

                                                                  ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### Firmware version
MK4duo - Firmware configurator tool (Rev 4.3.3) - Firmware Version 4.3.3

### My setup
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1620055/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1620058/Configuration_Pins.txt)

**Please be patient, we'll be trying to help you as soon as possible.**

Many thanks for your consideration.

Jon

iosonopersia, 2018-01-10:
Hi, from you configuration_overall it results that you have:
- Ramps-FD V2
- Eeprom on SD
- MKS OLED 1.3 128x64 SH1106 controller

The other settings are pretty straightforward. I'm going to check, but I feel that the problem is we do not have pins defined for your display in Ramps-FD v2 definition file...

Is it the first time you use MK4duo?

iosonopersia, 2018-01-10:
Yep, src/boards/1404.h does not define pins `DOGLCD_SCK`, `DOGLCD_MOSI`, `DOGLCD_CS ` and `DOGLCD_A0`. This is the source of the problem. We need to have a look at the pinouts of ramps-fd v2 and of MKS-12864OLED to find out which are the correct pins...

iosonopersia, 2018-01-10:
@jf1452 are you using this adapter? http://reprap.org/wiki/File:LCD-panel-adapter-FD-PINOUT.jpg

Hans007a, 2018-01-10:
Hello!

I use this adapter with DUO + RAMPS-FD v1 (a little security modded) with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and all works OK :-)

Hans007a, 2018-01-10:
Here is the MKS display:

https://sunhokeytech.en.alibaba.com/product/60603018493-803186233/Reprap_Ramps1_4_MKS_12864OLED_display_1_3inch_smart_controller_mini_oled_screen_oled_3d_printer_screen_1_3_Dots_LCD_with_cable.html

and it looks pin compatible to the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

jf1452, 2018-01-11:
Hello, yes, this is the display I would like to use with Ramps FD (moded to V2) with DUE processor board.  I traced the pins to be:

jf1452, 2018-01-11:
MKS_OLED_PIN, EXP1_PIN, ORIGINAL_LCD_FUNCTION, RAMPS-FD_AUX4_PIN, ARDUINO_PIN

MOSI, 3, E, 17, D17
CS, 4, RS, 18, D16
CLK, 5, D4, 16, D23
DC, 6, D5, 15, D25
RES, 7, D6, 14, D27

I guess you are only interested in the BOLD items.

jf1452, 2018-01-16:
@iosonopersia ; thanks for the picture of the board adapter; yes, this is the one I am using.
http://reprap.org/wiki/File:LCD-panel-adapter-FD-PINOUT.jpg

@Hans007a ; I tried looking for different board options including RAMPS-FD v1; but was still unable to compile.

I have tried a few combinations of boards in the hope that I can learn where to define DOGLCD pins to get this working, but so far no luck; still learning this build, and hope to be able to contribute one day; but until then, doing my best to learn what I can.

Many thanks

Jon

iosonopersia, 2018-02-27:
>MKS_OLED_PIN, EXP1_PIN, ORIGINAL_LCD_FUNCTION, RAMPS-FD_AUX4_PIN, ARDUINO_PIN

>MOSI, 3, E, 17, D17
CS, 4, RS, 18, D16
CLK, 5, D4, 16, D23
DC, 6, D5, 15, D25
RES, 7, D6, 14, D27

@jf1452 I took this as a reference, as I've now added MKS_12864OLED support in my repo. I need you to test if it does work. See this changes: #411