MKFirmware/MK4duo #370: BedLevel UBL compile error for HAL_DUE (v4.3.3)

dell9, 2018-01-10:
### Issue description
Not able to compile firmware when AUTO_BED_LEVELING_UBL enabled in configuration for Arduino Due (board id 1550, RuRAMPS4Due), IDE version 1.8.1

### Compile errors
`sketch\src/core/eeprom/eeprom.h:43:44: error: 'E2END' was not declared in this scope
         const static int16_t  meshes_end = E2END - 128; // 128 is a placeholder for the size of the MAT; the MAT will always`

after adding E2END definition to 1550.h the next error occures:
`sketch\src\feature\bedlevel\ubl\ubl_g29.cpp: In static member function 'static void unified_bed_leveling::manually_probe_remaining_mesh(const float&, const float&, const float&, const float&, bool)':
sketch\src\feature\bedlevel\ubl\ubl_g29.cpp:995:93: error: 'serialprintPGM' was not declared in this scope
         serialprintPGM(parser.seen('B') ? PSTR(MSG_UBL_BC_INSERT) : PSTR(MSG_UBL_BC_INSERT2));`

I've tried to replace serialprintPGM with SERIAL_MSG macro, and got the next error:
`sketch\src\lcd\ultralcd.cpp: In function 'void _lcd_ubl_map_lcd_edit_cmd()':

sketch\src\lcd\ultralcd.cpp:2353:74: error: 'dtostrf' was not declared in this scope

       dtostrf(pgm_read_float(&ubl._mesh_index_to_xpos[x_plot]), 0, 2, str);`

I've added #include <avr/dtostrf.h> to ultralcd.cpp and got OK compilation.

Not sure if all problems were fixed correctly...

### Firmware version
The actual 4.3.3

### My setup
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1620482/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1620483/Configuration_Pins.txt)

MagoKimbra, 2018-01-10:
I need information about how many bytes is the eeprom for UBL. I asked this to the developer.

dell9, 2018-01-11:
Contacted with board developer -  it has 256 kbit external EEPROM (24LC256T-I/SN chip, 8 banks with  32kbit each)

smoki3, 2018-01-19:
get the same on radds

sketch\src\feature\bedlevel\ubl\ubl_g29.cpp: In static member function 'static void unified_bed_leveling::manually_probe_remaining_mesh(const float&, const float&, const float&, const float&, bool)': sketch\src\feature\bedlevel\ubl\ubl_g29.cpp:995:93: error: 'serialprintPGM' was not declared in this scope serialprintPGM(parser.seen('B') ? PSTR(MSG_UBL_BC_INSERT) : PSTR(MSG_UBL_BC_INSERT2))

dell9, 2018-01-19:
**smoki3**,

try to change statement
serialprintPGM(parser.seen('B') ? PSTR(MSG_UBL_BC_INSERT) : PSTR(MSG_UBL_BC_INSERT2));

with 
SERIAL_MSG(parser.seen('B') ? MSG_UBL_BC_INSERT : MSG_UBL_BC_INSERT2);

in MK4duo\src\feature\bedlevel\ubl\ubl_g29.cpp

smoki3, 2018-01-19:
thanks!

i also had to add : #include <avr/dtostrf.h>

commit dfb7011d (2018-01-20): Fix #370
closed 2018-01-20 by MagoKimbra