MKFirmware/MK4duo #374: Auto Bed Leveling does not work.

VitDagger, 2018-01-14:
### Issue description
Installed BLTouch, I decided to configure Auto Bed Leveling. Configured according to the instructions.
**G28** runs well, with no errors. **G29** goes to the desired coordinates, takes a probe once and stops with an error "**Error: Probing failed**".
Board Ultratronics, drivers TMC2208, CoreXY.

Here is the extended log of the **G29** command
```
16:17:22.124 : G29
16:17:22.254 : mechanics.current_position=(50.000000, 50.000000, 16.399999) : >>> G29
16:17:22.254 : Machine Type: CoreXY
16:17:22.254 : Probe: BLTOUCH
16:17:22.254 : Echo:Probe Offset X:0.000000 Y:0.000000 Z:-1.400000 (Aligned With & Below Nozzle)
16:17:22.254 : Auto Bed Leveling: BILINEAR (disabled)
16:17:22.254 : DebugLevel:32
16:17:22.254 : mechanics.current_position=(50.000000, 50.000000, 16.399999) : probe.set_deployed
16:17:22.254 : deploy: 1
16:17:22.254 : raise(15.000000)
16:17:22.254 : >>> do_blocking_move_to(50.000000, 50.000000, 16.399999)
16:17:22.254 : <<< do_blocking_move_to
16:17:22.254 : mechanics.current_position=(50.000000, 50.000000, 16.399999) : bracket_probe_move
16:17:22.254 : >>> check_pt(50.000000, 50.000000, no stow)
16:17:22.254 : mechanics.current_position=(50.000000, 50.000000, 16.399999) :
16:17:22.254 : >>> do_blocking_move_to(50.000000, 50.000000, 16.399999)
16:17:22.254 : <<< do_blocking_move_to
16:17:22.254 : mechanics.current_position=(50.000000, 50.000000, 16.399999) : probe.set_deployed
16:17:22.254 : deploy: 1
16:17:22.254 : mechanics.current_position=(50.000000, 50.000000, 16.399999) : >>> run_z_probe
16:17:22.254 : mechanics.current_position=(50.000000, 50.000000, 16.399999) : >>> move_to_z
16:17:23.251 : set_bltouch_deployed(1)
16:17:23.251 : >>> do_blocking_move_to(50.000000, 50.000000, -10.000000)
16:17:37.447 : <<< do_blocking_move_to
16:17:37.447 : current_position=(50.000000, 50.000000, 1.643750) : sync_plan_position_kinematic
16:17:37.447 : mechanics.current_position=(50.000000, 50.000000, 1.643750) : <<< move_to_z
16:17:37.448 : >>> do_blocking_move_to(50.000000, 50.000000, 6.643750)
16:17:38.539 : <<< do_blocking_move_to
16:17:38.539 : <<< check_pt
16:17:38.539 : Error:Probing failed
16:17:38.539 : mechanics.current_position=(50.000000, 50.000000, 6.643750) : probe.set_deployed
16:17:38.539 : deploy: 0
16:17:38.539 : raise(15.000000)
16:17:38.539 : >>> do_blocking_move_to(50.000000, 50.000000, 16.399999)
16:17:40.279 : <<< do_blocking_move_to
16:17:40.279 : >>> do_blocking_move_to(50.000000, 50.000000, 16.399999)
16:17:40.279 : <<< do_blocking_move_to
16:17:40.279 : mechanics.current_position=(50.000000, 50.000000, 16.399999) : > probing complete
16:17:40.279 : mechanics.current_position=(50.000000, 50.000000, 16.399999) : bracket_probe_move
16:17:40.279 : <<< G29
16:17:40.279 : X:50.00 Y:50.00 Z:16.400 E:0.0000
```

Here is the video of the process G28 and G29 https://youtu.be/HdVMh6JR9c8

### Firmware version
Version 4.3.3 Stable

### My setup
#define Z_ENDSTOP_SERVO_NR 0
#define Z_ENDSTOP_SERVO_ANGLES {10,90} // Z Servo Deploy and Stow angles
#define BLTOUCH
#define BLTOUCH_DELAY 375 // (ms) Enable and increase if needed
#define X_PROBE_OFFSET_FROM_NOZZLE  29     // X offset: -left  [of the nozzle] +right
#define Y_PROBE_OFFSET_FROM_NOZZLE  -10     // Y offset: -front [of the nozzle] +behind
#define Z_PROBE_OFFSET_FROM_NOZZLE -1.4     // Z offset: -below [of the nozzle] (always negative!)
#define Z_PROBE_SPEED_FAST 1000
#define Z_MIN_PROBE_REPEATABILITY_TEST
#define AUTO_BED_LEVELING_BILINEAR
#define LEFT_PROBE_BED_POSITION 50
#define RIGHT_PROBE_BED_POSITION 250
#define FRONT_PROBE_BED_POSITION 50
#define BACK_PROBE_BED_POSITION 150
#define NUM_SERVOS 1
#define Z_PROBE_PIN         ORIG_Z_PROBE_PIN

MagoKimbra, 2018-01-14:
From that mistake if he finds that the probe did not retract well or did not extend well. Check that it works perfectly sometimes it crashes.

VitDagger, 2018-01-14:
There are no problems with this. It works for G28 and during the print. The problem exists only for G29.
It retracts and extends well

dell9, 2018-01-22:
Try to execute G29 V4 - it will produce more detailed log. May be it will be helpful for investigation