MKFirmware/MK4duo #376: [4.3.3] AUTO_FAN is only triggered by steppers

Nume1977, 2018-01-15:
I have BOARD_MKS_13 board, and I wanted to enable AUTO_FAN when hotend 0 reached a certain threshold, so I had to change ORIG_FAN1_PIN to 7 enable the use of the FAN1 mosfet.
I left the original settings on #define AUTO_FAN { -1, 0, -1, -1, -1, -1 }, but instead of triggering when the hotend heats up/down it always triggers when the stepper are on (same thing as putting 7) and turns off 60 seconds after they are powered down!

MagoKimbra, 2018-01-15:
Have you the eeprom?
These values are saved in eeprom, perhaps you have saved with incorrect values. Use the M106 command to set the right hotend for the fan.
First of all it resets
M106 P1 H-1
then
M106 P1 H0

Nume1977, 2018-01-15:
That did the trick! But it is strange because i resetted the EEPROM from the menu...

MagoKimbra, 2018-01-15:
Ok it's equal.. But in this mode you have resetted all EEPROM..

smoki3, 2018-01-20:
I can't get autofan to work. I use the default define for my radds.

iosonopersia, 2018-08-04:
@smoki3 if you still suffer from that problem, please open a new PR. Thanks.

closed 2018-08-04 by iosonopersia