MKFirmware/MK4duo #377: Configurator fails when using values with decimal places

Nume1977, 2018-01-15:
If a value is set using a decimal place (ex: 1.05 / 326.96) the online configurator will fail!

This is a copy / paste of the return address, which shows the php error line:
"warning: Invalid argument supplied for foreach() in /customers/5/9/7/marlinkimbra.it/httpd.www/configurator/v4_3_3/config.php on line 1558../tmp/builder_1516007314-5a5c6f9227330/zip/MK4duo_15-01-2018_100834.zip"

MagoKimbra, 2018-01-15:
Use the comma not the point.

Nume1977, 2018-01-15:
Thank you for the information! I think it would be important to have that information on the page, or even better lock the inputs to only accept commas!

MagoKimbra, 2018-01-15:
But this is a function of the browser, that is, it is normal that in a browser when asking for float numbers, use the comma. It's not something I did, they are html fields.

Nume1977, 2018-01-15:
Maybe you can tweak the php code to accept both options ( . and , )

MagoKimbra, 2018-01-18:
I do not know a lot about php, you know how?

Nume1977, 2018-01-18:
The most simple way would be:
$output = str_replace(",", ".", $your_input); // Replace , with .
or
$output = str_replace(".", ",", $your_input); // Replace . with , 

I can take a look at the html code of the site and see if it some javascript can be implemented to validate the fields and/or only accept formatted values.