MKFirmware/MK4duo #380: scroll of long names on SD card don't works

Hans007a, 2018-01-16:
Hello!

scroll of long names on SD card don't works with due + RAMPS-FD v1 + RRFGD

Regards

Hans

Nume1977, 2018-01-17:
Confirmed. 
MKS Base 13 + Reprap discount LCD

MagoKimbra, 2018-01-17:
What do you mean by scroll long names does not work?

Nume1977, 2018-01-17:
On 128x64 LCD the file name is truncated to 20 characters long, with this option the filename should scroll <-> sideways and display all of the text (it works on marlin 1.1.8 i just tested it), but it does nothing on Kimbra!

Hans007a, 2018-01-17:
I mean this:
```
// Enable this option to scroll long filenames in the SD card menu
#define SCROLL_LONG_FILENAMES
```

Nume1977, 2018-01-17:
Yes, that option has no effect on Kimbra, I have it on, but no scroll!

MagoKimbra, 2018-01-17:
A ok!! Check

Hans007a, 2018-01-17:
And what is with my other issu with the wrong display of top bed temperature?

commit 1cec6cf2 (2018-01-17): Fix #380
closed 2018-01-17 by MagoKimbra