MKFirmware/MK4duo #382: Movements / command procrssing bugs in v4.3.3

dell9, 2018-01-20:
Hello. Recently updated firmware from 4.3.29_dev version to actual 4.3.3 and found some bugs.

If print pause is requested from LCD menu with subsequent print stop action requesting, then no stop is actually done, just pause. If stop is selected with no pause - all works correctly.

The second issue - if printing is stopped already and I moves Z down by 10mm, Y begins moving too in zero point direction and far away from zero point. 

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1648670/Configuration_Overall.txt)

MagoKimbra, 2018-01-20:
I do not understand... Please explain your problem better!!!

dell9, 2018-01-20:
**The first issue.**
I select file to be printed from SD card, printing started successfully.
I go to 12864 LCD menu and select Pause, printing paused successfully
I go to 12864 LCD menu again and select Stop. Nothing happened - bed is still heating, hotend the same, progress bar on LCD shows printing status.

**The second issue.**
I select file to be printed from SD card, printing started successfully.
I go to 12864 LCD menu again and select Stop, printing stopped successfully.
I go to 12864 LCD menu and select MOVE Z by 10mm down. Z is moving but in the same time Y is moving too to zero direction. Y reaches zero point and trying to move more and more. After near 5sec Y moving is stopped.

MagoKimbra, 2018-01-20:
Try now..

For problem with second issue. Now when you stop, save the position for the restart and move from the piece, this for a possible restart. But you can not move if not before a possible Home. I see if I can find a cleaner solution, but for now you can not move after the stop if not doing a Home first.

iosonopersia, 2018-03-16:
@dell9 does it work now? Have you still this problems? Please make us know

dell9, 2018-03-16:
Pause worked correctly. Waiting solution for bug with movement after print stop...