MKFirmware/MK4duo #385: MKS OLED not defined for DUE & RAMPS-FD

jf1452, 2018-01-23:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
I have been able to configure and compile 4.3.4 for DUE + RAMPS-FD (v2) with Full graphic display and also Nextion 2.8". Unfortunately, the configurator still does not support MKS OLED display.

Previously reported in #369 

### Compile errors
Errors reported look very similar to 4.3.3.

Arduino: 1.8.5 (Windows 10), Board: "Arduino Due (Native USB Port)"

In file included from sketch\src\lcd\ultralcd.cpp:64:0:

sketch\src\lcd\ultralcd_impl_DOGM.h:204:30: error: 'DOGLCD_SCK' was not declared in this scope

     U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

                              ^

sketch\src\lcd\ultralcd_impl_DOGM.h:204:42: error: 'DOGLCD_MOSI' was not declared in this scope

     U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

                                          ^

sketch\src\lcd\ultralcd_impl_DOGM.h:204:55: error: 'DOGLCD_CS' was not declared in this scope

     U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

                                                       ^

sketch\src\lcd\ultralcd_impl_DOGM.h:204:66: error: 'DOGLCD_A0' was not declared in this scope

     U8GLIB_SH1106_128X64 u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);      // 8 stripes

                                                                  ^

exit status 1
Error compiling for board Arduino Due (Native USB Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
Latest 4.3.4

### My setup

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1657435/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1657436/Configuration_Pins.txt)

**Please be patient, we'll be trying to help you as soon as possible.**

jf1452, 2018-01-23:
Thank you very much for 4.3.4.

I really like the new support for the Arduino DUE native port (I broke my programming port); also Native programs much faster. This saves me the manual edit of the code to use the port.

I also very much appreciate the stepper drive delay option (us); this seems to have helped overcome a glitch when driving on DRV8825 which occasionally lost some steps.

Cheers

Jon

jf1452, 2018-02-18:
Just tried this configuration option using 4.3.5; but unfortunately, the pins are still not configured.

Can someone help me by identifying where to add the definitions for these pins?

iosonopersia, 2018-02-27:
@jf1452 I've opened a PR to add support for MKS 12864 OLED display to Ramps FD v2.
Could you please try #411 changes to see if it does work? Thanks

iosonopersia, 2018-03-03:
@jf1452 up

closed 2018-08-25 by iosonopersia