MKFirmware/MK4duo #386: Due + RURAPMS4D, Heater2 does not turn on

svs072477, 2018-01-30:
MK4Duo 4.3.3, 4.3.4
Due + RURAPMS4D, Heater2 (on pin 11) does not turn on

MagoKimbra, 2018-01-30:
Put M306 H1 and read pin for Heater 2.

svs072477, 2018-01-30:
Sorry, not really right at once wrote it was about heter3 (H2)
[image]

MagoKimbra, 2018-01-30:
Exusme... How many heaters do you have, Hotend?

svs072477, 2018-01-30:
Three hotends. On pins 13, 12, 11 like on raddds
//###HEATER
#define ORIG_HEATER_0_PIN          13
#define ORIG_HEATER_1_PIN          12
#define ORIG_HEATER_2_PIN          11
#define ORIG_HEATER_3_PIN          NoPin
#define ORIG_HEATER_BED_PIN         7
#define ORIG_HEATER_CHAMBER_PIN     6
#define ORIG_COOLER_PIN            NoPin

With marlin 2.0 all three work

MagoKimbra, 2018-01-30:
Yes, but Marlin 2.0 not use hardware PWM, mK yes.
// Heater outputs
// Bed PMW: D7 has hardware PWM so bed has PWM
// h0, h1 PMW: D13 & D12 are on TIOB0 & B8 which are both TC B channels, so they get PWM
// h2 bang-bang: D11 is on TIOA8 which is a TC A channel shared with h1, it gets bang-bang control
Sorry but this is a limit of radds. Put PID off on H2 and use bang bang mode on Heater 2

svs072477, 2018-01-30:
Thank you explained

closed 2018-01-30 by svs072477