MKFirmware/MK4duo #387: Unable to flash mk4duo_2_4_basic_433.tft

jf1452, 2018-02-01:
### Issue description
I have copied the mk4duo_2_4_basic_433.tft to a SD card and tried to Flash my nextion LCD. I get the error on the Nextion display:

Check Data... 100%
Update Failed:check Error!

### Compile errors
No problems with compilation

### Firmware version
4.3.4

### My setup
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1686544/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1686545/Configuration_Pins.txt)

**Please be patient, we'll be trying to help you as soon as possible.**

iworld4us, 2018-02-07:
Hello. Please check the following things:
- First of all, make sure that your display model is **exactly** `NX3224T024`.
- Also, sometimes you need to try to flash the firmware one more time and it succeeds.

jf1452, 2018-02-18:
Thank you iworld4us.

My display is NX3224T024_011; I have no problems flashing the Nextion demo onto the HMI display using USB/serial, or by SD card. Device flashes first time everytime.

I hoped that the new 4.3.5 firmware would overcome my problem; but I am still getting the same error as before.

Still not sure what to try next.

Thanks

iworld4us, 2018-02-18:
Hi, please try out new Nextion FW version: [mk4duo_2_4_basic_434.tft](https://github.com/iworld4us/nextion-2.4/blob/master/mk4duo_2_4_basic_434.tft).

jf1452, 2018-02-19:
Hello iworld4us; yes, the new firmware installs correctly; many thanks for this, you have resolved the problem :).

closed 2018-02-19 by jf1452