MKFirmware/MK4duo #390: Compiling Errors with LCD ( PanelOne )

Aracos, 2018-02-03:
Issue description
Can´t compile with defined LCD ( PanelOne ) 
Compiling for RAMPS4DUE

Compile errors
`In file included from C:\Users\Ironaxe\AppData\Local\Temp\arduino_build_687509\sketch\src/HAL/HAL.h:39:0,

                 from C:\Users\Ironaxe\AppData\Local\Temp\arduino_build_687509\sketch\MK4duo.h:107,

                 from C:\Users\Ironaxe\Desktop\MK4DUE\OLD VERSION LCD\MK4duo\MK4duo.ino:294:

C:\Users\Ironaxe\AppData\Local\Temp\arduino_build_687509\sketch\src/core/commands/parser.h: In static member function 'static char* GCodeParser::temp_units_name()':

C:\Users\Ironaxe\AppData\Local\Temp\arduino_build_687509\sketch\src/core/commands/parser.h:247:127: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

           return input_temp_units == TEMPUNIT_K ? PSTR("Kelvin") : input_temp_units == TEMPUNIT_F ? PSTR("Fahrenheit") : PSTR("Celsius");

                                                                                                                               ^

C:\Users\Ironaxe\AppData\Local\Temp\arduino_build_687509\sketch\src/HAL/HAL_DUE/HAL_Due.h:86:17: note: in definition of macro 'PSTR'

 #define PSTR(s) s

                 ^

Bibliothek LiquidCrystal in Version 1.0.7 im Ordner: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\libraries\LiquidCrystal  wird verwendet
Bibliothek Wire in Version 1.0 im Ordner: C:\Users\Ironaxe\Documents\ArduinoData\packages\arduino\hardware\sam\1.6.11\libraries\Wire  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).`

### Firmware version
4.3.4 / 4.3.3 stable

My setup
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1691847/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1691848/Configuration_Pins.txt)

THX for having a look ^^

Aracos, 2018-02-04:
Just found out that i had to commen out this line in Config_Overall:

//#define TEMPERATURE_UNITS_SUPPORT

then it compiles fine.
( Display does not work though, maybe it is broken. got to test it )

iosonopersia, 2018-03-16:
@Aracos I wasn't able to reproduce this problem... does it work now for you? Please make us know