MKFirmware/MK4duo #392: M303: temperature keeps rising to max

iworld4us, 2018-02-07:
### Firmware version
- `Version 4.3.4 Stable`
- Arduino DUE

### Issue description
When I run the following GCODE:
```gcode
M303 H0
```

When `cycles` variable in `Temperature::PID_autotune`:
https://github.com/MKFirmware/MK4duo/blob/c12ca9818a14d97b9f5e52a554b503e5c04f44c7/MK4duo/src/core/temperature/temperature.cpp#L395

becomes`1`, the heater temperature either keeps rising to maximum or locks forever.

I think that the problem is in line:
https://github.com/MKFirmware/MK4duo/blob/c12ca9818a14d97b9f5e52a554b503e5c04f44c7/MK4duo/src/core/temperature/temperature.cpp#L441

MagoKimbra, 2018-02-07:
But this is the routine for the calculation of the pid, it is normal that it continues to rise, then there is control if the current temperature has exceeded the set value ..

iworld4us, 2018-02-08:
when set temp > 140  then ( bias + d ) > 254

MagoKimbra, 2018-02-08:
No!! soft_pwm is the pwm value not temp. This value is used exclusively to drive the duty cycle of the PWM signal.

iworld4us, 2018-02-08:
Yes, but    'PID Autotune failed! timeout' when ( bias + d ) > 254.  ?

MagoKimbra, 2018-02-08:
No, that's not the problem. It goes into timeout because it can not find the right values for the pid.
How much do you feed the card? 12 or 24?

iworld4us, 2018-02-08:
24V    30W   2 x UM2 heads

MagoKimbra, 2018-02-08:
Ok, the resistance for the nozzle is 24 v?

iworld4us, 2018-02-08:
24v 10 ohm 30w

iworld4us, 2018-02-08:
24v 20 Om 30w

iworld4us, 2018-02-10:
I reduced MOSFET optical key voltage to **12 V**, now timeouts happen when temperature is set to bigger than `195`.
Before with **24 V** it was happening with temperature bigger than `120`.

MagoKimbra, 2018-02-10:
Very strange I have the hotend to 24 and not a problem ... I do not understand why you do not get to do the pid ... You can send me the hole of repetrier with the graph of the temperatures during the autotune and also the log if you can ...

iworld4us, 2018-02-10:
MOSFET 55NF06L  
R1 = 100k

[image]

Temperature graph:

[image]

iworld4us, 2018-02-12:
Reducing hot end fan speed solved this issue!

closed 2018-02-12 by iworld4us

MagoKimbra, 2018-02-12:
Ok!! THX!