MKFirmware/MK4duo #396: Problems with Endstop Pullup

CltAvi8tor, 2018-02-09:
### Issue description
Using this on an Arduino Due with a modified RAMPS v1.4. My old build of Marlin for the laser works. When using this build, the endstop pullup is not working properly; with the original version I see a voltage go from +5VDC to zero when the endstops are hit. With this version the voltage drops to ~ 3.2VDC. 

I have tried variations of the code:
#define ENDSTOPPULLUP_XMIN    true

but cannot get this to work. Everything else seems to be working except this. Any help greatly appreciated.

### Firmware version
V4.3.4

### My setup
Arduino Due
RAMPS v1.4
K40 laser

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
#define ENDSTOPPULLUP_XMIN    true

iosonopersia, 2018-02-09:
In MK4duo v4.3.4 the endstop pullup settings are stored inside the EEPROM. This means you can change them without having to recompile everything. But it also mean that you'll have to start your printer, send a M500 command using a host software, then turn off and restart the printer.
This way, you will save the settings of your printer inside the EEPROM chip. You'll have to restart the printer because updating pullup settings can only done during the initial boot of the firmware, not while the printer is already running.

Let us know...

CltAvi8tor, 2018-02-09:
Thanks - testing now.

FWIW, the docs are incorrect with regards to the gcode; this is from the Gcodes.md file (note the two M122 entries - this should be 122, 123, and 124):

**M122** | MIN_SOFTWARE_ENDSTOPS or MAX_SOFTWARE_ENDSTOPS | S<1=true/0=false> Enable or disable check software endstop
-- | -- | --
**M122** | ENDSTOP | Set Endstop Logic X Y Z I J K P D F W
**M123** | ENDSTOP | Set Endstop Pullup X Y Z I J K P D F W

MagoKimbra, 2018-02-09:
Ops sorry i fix it!!

CltAvi8tor, 2018-02-09:
No worries - THANK YOU, and I personally appreciate your work. I've spent a LOT of time with the old Marlin that this was based on and made a lot of customization, and now I'm trying to understand your build. I love the structure and organization, excellent work.

And FWIW the use of stored EEPROM settings didn't work - I think something is up with the pin mapping for the XYZ endstops, I keep getting a floating voltage of around 3,2VDC when the sensor is set as opposed to 0VDC with the original firmware. Any ideas where to look for this?

Thanks again!!!

CltAvi8tor, 2018-02-14:
Okay, a bit more here - I finally got this working, but I think there may be a logic problem. Even though I've got the pullup set via Configuration_Cartesian.h using:

#define ENDSTOPPULLUP_XMIN    false
#define ENDSTOPPULLUP_YMIN    false

It would not work, no matter the combinations of settings. However, changing endstops.cpp to the below it works (I added the pullup outside the if to force it to set):

  #if HAS_X_MIN
    if (Is_pullup(X_MIN))
      SET_INPUT_PULLUP(X_MIN_PIN);
    else
      SET_INPUT(X_MIN_PIN);
		// This is the change I made
		SET_INPUT_PULLUP(X_MIN_PIN);
  #endif

Suggestions?

Thanks!

MagoKimbra, 2018-02-14:
Sorry but so always puts the pullup if it is true that false!

CltAvi8tor, 2018-02-14:
Yes - I know that, I'm forcibly doing this to *explicitly* ensure that the pin is being set as a pullup enabled input. My point is that the *logic* in the code itself is broken - setting the correct values for the endstop pullups should do the same thing but it's not...

MagoKimbra, 2018-02-14:
But if you put #define ENDSTOPPULLUP_XMIN false there is no pullup ..
Remember that the values are saved in eeprom, so if you change from configuration it remains the one you had previously set. Change it to M124 X1 and M500 to save in eeprom.
Restart the board annd check !!!

CltAvi8tor, 2018-02-14:
Okay, let me understand this, per the code, uncommenting that line is supposed to enable pullups (see below). Are you saying that commenting the lines will *enable them? Or would the correct setting be to leave them uncommented but set to true? I appreciate the help here (and I've tried, I thought, every combination of this).

/*****************************************************************************************
 ************************* Endstop pullup resistors **************************************
 *****************************************************************************************
 *                                                                                       *
 * Comment this out (using // at the start of the line) to                               *
 * **disable** the endstop pullup resistors                                                  *
 *                                                                                       *
 *****************************************************************************************/
#define ENDSTOPPULLUP_XMIN    false
#define ENDSTOPPULLUP_YMIN    false

MagoKimbra, 2018-02-14:
A ok.. The comments are left with the old system to disable or enable the define .. You're right ... I edit the part of the comments ...

bekir86, 2018-02-22:
Hello  May I ask questions about endstops ?  First I want to say I don't know about codes. 
I configurated 4.3.2 . it was fine. I added endstops It didnt work. I found the problem, when I pushed trigger , Axis was moving so I changed logic. it didn't solve problem . Also I want to sent X axis to max limit  I changed "-1" to "1" . This time  Axis didn't move while I was pushing trigger.  I tried other versions 4.3.5  The same problem. :(

MagoKimbra, 2018-02-22:
Please send M119 in command on HOST and read the log, with last firmware you can change the logic with command gcode...

bekir86, 2018-02-22:
Yes I did it  know I am testing firmwire so I got only x min endstop  configurations: 
#define ENDSTOPPULLUP_XMIN    true
#define ENDSTOPPULLUP_YMIN    true
#define ENDSTOPPULLUP_ZMIN    true
--------------------------------
#define X_MIN_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_LOGIC   false   // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_LOGIC   true   // set to true to invert the logic of the endstop.

result
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
z_min: TRIGGERED

closed 2018-08-04 by iosonopersia