MKFirmware/MK4duo #399: PID Autotune :Error writing to EEPROM!

iworld4us, 2018-02-19:
v3.3.5
Arduino Due EEPROM 24lc64 0x2000
Brightness setting stored Ok!

>>> m303 u7
SENDING:M303 U7
PID Autotune start
Hotend:0 Temp:200 Cycles:5 Method:0 Apply result
busy:heating
...
busy:heating
PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration!
 Kp: 11.499966 Ki: 0.631866 Kd: 52.324844
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Error writing to EEPROM!
Echo:Settings Stored (22 bytes; crc 34772)
busy:paused for user

MagoKimbra, 2018-02-19:
Arduino due with EEPROM I2C or SPI??

iworld4us, 2018-02-19:
[image]

iworld4us, 2018-02-21:
#define EEPROM_FLASH
solve problem, but for v4.3.5 it must be uncommented in configurator!

closed 2018-02-21 by iworld4us