MKFirmware/MK4duo #403: Prints randomly pausing while printing and then resuming in a couple of minutes

Beherith, 2018-02-21:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
I updated mk4duo on my DUE+Hacked RAMPS+TMC2130 from 4.3.3 to 4.3.5 by pulling the latest github repo, and transferring diffing the configuration.h files in the old version and the new one and transferring over settings. 

After the update, whenever I print anything, even small objects with print times less than 30 minutes, approximately every 5-10 minutes, the printer pauses for a minute, with the following output on the serial terminal:

Printing does resume after about 2 minutes, but the print is ruined by the hotend. 

Send: G1 X84.657 Y44.285 E95.17268
Recv: ok
Send: G1 X84.725 Y44.365 E95.17488
Recv: ok
Send: G1 X84.651 Y44.667 E95.1814
Recv: ok
Send: G1 X84.266 Y46.395 E95.21849
Recv: ok
Send: G1 X84.105 Y47.001 E95.23163
Recv:  T:235.05 /235 B:60.14 /60 B@:81 @:72
Recv:  T:235.17 /235 B:60.15 /60 B@:84 @:68
Recv:  T:234.94 /235 B:60.14 /60 B@:73 @:65
[deleted 100 lines]
Recv:  T:235.63 /235 B:60.01 /60 B@:84 @:60
Recv:  T:235.46 /235 B:59.97 /60 B@:77 @:64
Recv:  T:235.28 /235 B:59.97 /60 B@:96 @:69
Recv: ok
Send: G1 X83.806 Y47.907 E95.25162
Recv: ok
Send: G1 X84.318 Y48.128 E95.26331
Recv: ok
Send: G1 X84.409 Y48.156 E95.2653

### Compile errors
No compile issues

### Firmware version
MK4duo's versions that have this issue: 4.3.5

### My setup

[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1744428/Configuration_Pins.txt)
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1744429/Configuration_Overall.txt)

Beherith, 2018-02-21:
I also tested printing from SD card and via octoprint. The same random pausing happens. I can interact with the LCD while it is paused, so I dont know what the firmware is waiting for :/

Beherith, 2018-02-23:
Recv: busy:processing

After enabling some debug options, the firmware sends busy:processing messages while doing nothing.

MagoKimbra, 2018-02-23:
It's normal, if you do not want messages or your host from problems you can disable them:

```
/**
 * Host Keepalive
 *
 * When enabled MK4duo will send a busy status message to the host
 * every couple of seconds when it can't accept commands.
 */
// Disable this if your host doesn't like keepalive messages
#define HOST_KEEPALIVE_FEATURE
// Number of seconds between "busy" messages. Set with M113.
#define DEFAULT_KEEPALIVE_INTERVAL 2
// Some hosts require "busy" messages even during heating
#define BUSY_WHILE_HEATING
```

Beherith, 2018-02-23:
Yeah, I know I can disable that, I enabled it specifically to find out why the printer is just pausing in the middle of a print. Im trying some runs with M111 S7

Beherith, 2018-02-24:
So some serial corruption seems to be going on, I initially thought that it was an USB communication error (using the programming port on the Due), because I got a few messages like this:

018-02-23 17:35:36,985 - Send: M204 S600
2018-02-23 17:35:36,988 - Recv: Echo:Unknown command: "k"
2018-02-23 17:35:36,989 - Recv: ok
2018-02-23 17:35:36,991 - Send: G0 F6000 X117.376 Y112.857
2018-02-23 17:35:36,996 - Recv: Echo:Unknown command: "F2400 X116.819 Y113.979 E249.63491"
2018-02-23 17:35:36,997 - Recv: ok
2018-02-23 17:35:36,999 - Send: M204 S800
2018-02-23 17:35:37,003 - Recv: Echo:Unknown command: "400 X116749 Y114.049"

But then I got some messages like this one:

2018-02-23 18:09:42,629 - Send: M204 S800
2018-02-23 18:09:42,634 - Recv: ok
2018-02-23 18:09:42,637 - Send: M205 X5 Y5
2018-02-23 18:09:42,641 - Recv:  T:261.84 /260 B:100.11 /100 B@:151 @:46
2018-02-23 18:09:42,653 - Recv: Setting Print and Travel acceleration: 800.000000
2018-02-23 18:09:42,654 - Recv: ok
**2018-02-23 18:09:42,659 - Recv: Echo:Unknown command: "eleratioY5"**
2018-02-23 18:09:42,660 - Recv: ok
2018-02-23 18:09:42,662 - Send: G1 F2400 X99.606 Y105.175 E2636.40839
2018-02-23 18:09:42,664 - Send: G1 X99.721 Y104.927 E2636.41748

and

2018-02-23 18:02:38,634 - Recv:  T:261.48 /260 B:100.10 /100 B@:158 @:62
2018-02-23 18:02:38,647 - Recv: Setting Print and Travel acceleration: 800.000000
2018-02-23 18:02:38,649 - Recv: ok
2018-02-23 18:02:38,661 - Send: G1 F2400 X97.422 Y104.126 E2187.59131
**2018-02-23 18:02:38,665 - Recv: Echo:Unknown command: "and TravY5"**

I checked my gcode file, and the word acceleration does not occur in it, so the only way that the firmware could have got "eleratioY5" as an input, if it mixed up the serial input/output buffers somehow.

Beherith, 2018-02-26:
How should I go along debugging the serial buffer corruption, as the more I look at it, the more the echo'd commands look like they are actually something that was previously in the send buffer.
2018-02-26 10:06:49,430 - Recv: Echo:M204 S800
2018-02-26 10:06:49,434 - Recv: Setting Print and Travel acceleration: 800.**000000
2018-02-26 10:06:49,435 - Recv: ok**
2018-02-26 10:06:49,437 - Send: G1 F6000 E61.87468
2018-02-26 10:06:49,439 - Recv: Echo:0000
2018-02-26 10:06:49,442 - **Recv: Echo:Unknown command: "0000"**
2018-02-26 10:06:49,444 - Recv: ok
2018-02-26 10:06:49,446 - Send: G0 F2700 X91.537 Y184.728
2018-02-26 10:06:49,448 - Recv: Echo:ok Y10
2018-02-26 10:06:49,451 - **Recv: Echo:Unknown command: "ok Y10"**
2018-02-26 10:06:49,452 - Recv: ok

johannes-otto, 2018-02-26:
I got the same problem.

iosonopersia, 2018-02-26:
I feel like this is related to https://github.com/MarlinFirmware/Marlin/issues/9111

What BAUDRATE are you using?

Beherith, 2018-02-26:
I am using 250000, as that is a perfect match for both the SAM and the atmega16u clocks. 

@johannes-otto, are you using octoprint? Do you experience the same issues with SD card printing as well?

Beherith, 2018-02-27:
So I rigged up an stm32 to log both serial tx and rx on the due, patiently waiting for it to crash so i can ascertain whether its code or just unstable serial.

johannes-otto, 2018-02-27:
I'm also using 250000. I currently use a due with a smart ramps board. I also have TMC2130 with soft spi.
I cant print from SD at the moment, since enabeling sdsupport crashes my display. So I use only Octoprint.

Beherith, 2018-02-27:
Note that I made MK4DUO echo back any command that is entering the Command Queue at 
https://github.com/MKFirmware/MK4duo/blob/master/MK4duo/src/core/commands/commands.cpp#L206
to:
```
      // Add the command to the queue
      SERIAL_TXT("Enqueued command: ");
      SERIAL_TXT(serial_line_buffer);
      SERIAL_EOL();
      enqueue_command(serial_line_buffer, true);
```

So in my logging of serial TX and RX on the due with another microcont I found some reeeeally wierd stuff. Where 2: is (atmega16u serialusb)->Due and 1: is Due->(atmega16u serialusb)

2: **Enqu** G1 X87.968 Y182.116 E11.78357*109
1: Echo:N103 G1 X87.505 Y183.08 E11.73274*88
1: ok
1: Echo:N104 G1 X87.7 Y182.581 E11.75813*96
1: ok
1: Enqueued command: **Enqu** G1 X87.968 Y182.116 E11.78357*109
1: Echo:**Enqu** G1 X87.968 Y182.116 E11.78357*109
1: Echo:Unknown command: "**Enqu** G1 X87.968 Y182.116 E11.78357"
1: ok
2: N106 M105*32
1: Line number is not good: N106 M105*32Error:Line Number is not Last Line Number+1, Last Line: 104
N1: Resend:105

In the above, the HOST seems to be sending the chars **Enqu** to the due, which is impossible, as the host doesnt ever send any commands like that.

Also, in the next case, they HOST seems to be sending **Echo**, which is also impossible, as octoprint never says echo.

1: Echo:N11915 G1 X127.942 Y184.509 E428.50798*104
2: Echo:N1 X128.272 Y184.569 E428.51434*110
1: ok
1: Enqueued command: **Echo**:N1 X128.272 Y184.569 E428.51434*110
1: Echo:**Echo**:N1 X128.272 Y184.569 E428.51434*110
1: Echo:Unknown command: "**Echo**:N1 X128.272 Y184.569 E428.51434"
1: ok
2: N11917 G1 X128.498 Y184.892 E428.52181*108
1: Line number is not good: N11917 G1 X128.498 Y184.892 E428.52181*108Error:Line Number is not Last Line Number+1, Last Line: 11915
1: Resend:11916
1: ok

I compared these parts with the actual octoprint serial logs
2018-02-26 23:25:36,647 - Send: N11917 G1 X128.498 Y184.892 E428.52181*108
2018-02-26 23:25:36,654 - Recv: Enqueued command: Echo:N1 X128.272 Y184.569 E428.51434*11N11917 GEcho:N1 X128.272 Y184.569 E428.51434*110
2018-02-26 23:25:36,659 - Recv: Echo:Unknown command: "Echo:N1 X128.272 Y184.569 E428.51434"
2018-

Here is the sketch for the dual serial logger:

`
const int baudrate = 250000;

char buf1[256];
int bptr1 = 0;
char buf2[256];
int bptr2 = 0;
unsigned long t = 0;
void setup() {
  // put your setup code here, to run once:
  
  Serial1.begin(baudrate);
  Serial2.begin(baudrate);
  Serial.begin(baudrate);
  for (int i = 0; i<10;i++){
    Serial.println(i);
    delay(1000);
  }

  

}

void loop() {
  // put your main code here, to run repeatedly:
  while (Serial1.available()){
    byte val = (byte) Serial1.read();
    if (bptr1 <256){
      buf1[bptr1] = val;
      bptr1 ++;
    }else{
        bptr1 = 0; //die die die 
    }
    if (val == '\n'){
      Serial.print("1: ");
      Serial.print(buf1);
      bptr1 = 0;
    }
    
    }
  while (Serial2.available()){
    byte val = (byte) Serial2.read();
    if (bptr2 <256){
      buf2[bptr2] = val;
      bptr2 ++;
    }else{
        bptr2 = 0; //die die die 
    }
    if (val == '\n'){
      Serial.print("2: ");
      Serial.print(buf2);
      bptr2 = 0;
    }
    }
  if ( millis() - t> 1000){  
    Serial.println(t);
    t = millis();
  } 
}
`

MagoKimbra, 2018-02-27:
Mk4duo on board with DUE processor uses the serial hardware contained in the variants provided by Arduino, the firmware reads the buffer that is managed by these routines, if that were the way you say, then it would be the library to have the problem. But I use a card with the DUE and I have never encountered any problems on the serial.
However, I verify by doing as you did. on my board ultratronics...

Beherith, 2018-02-27:
So I got it to completely fail, and according to my custom hardware serial logger, the command 
1: Resend:2886
Was sent by the Due.
Octoprint recieved:
2018-02-27 10:34:49,800 - Recv: Resend:28N2885 G1 Error:Line Number is not Last Line Number+1, Last Line: 2885
2018-02-27 10:34:49,802 - Printer requested line 28 but no sufficient history is available, can't resend

I tried driving the due serial from an external FTDI usb->serial adapter on pins D0 and D1, but while I can perfectly read it as usually, i cant seem to write to the RX pin of the Due, because i seem to have a crappy chinese clone of the board, which means that ENABLE line of the 74LVC1G125 buffer between the ATMega16U TX pin and the Due RX pin is always on. Ergo I cant drive that unless I cut a trace :(

Any other ideas to reduce the number of errors? I previously tried using the native USB port of the Due as an interface, but errors there resulted in dropped connections, which are even worse than just corrupt packets. I have tried every single micro USB cable in the house, even added some ferrite beads to them, to no avail :(

MagoKimbra, 2018-02-27:
You can try to connect Raspberry directly with the DUE using the pins (both 3.3v) bypassing the usb and see what happens.

Beherith, 2018-02-27:
I hope yall dont mind me using this ticket as a blog, but anyhow, i'm going to rip out the bufferin front of the serial tx and add a regular ftdi usbserial. Or i could just try lower baud rates but that would be admitting defeat

Beherith, 2018-02-27:
Ah @magokimbra, you mean connect raspi serial straight to due serial? The reason i cant do that is that the voltage buffer between atmega16tx and due rx is always on :(

johannes-otto, 2018-02-28:
I can confirm now, that lowering the Baudrate to 115200 let me print 2 Benchys without a random pause.

Beherith, 2018-02-28:
I went down to 115200 as well, yet I still get a very large number of Resend requests - and I have proven those to still be an issue.
Trying an FDTI chip on Serial1 now.

johannes-otto, 2018-02-28:
@Beherith 
apart from the resend requests, can you print @115200 ? Not that it deligitimates the issue, but I got some projects to print.

Beherith, 2018-02-28:
@johannes-otto Well, since I put an FTDI chip on Serial1, and am still getting the same issues, it may just be a raspi usb driver bug (apparrently they are pretty bad), which may also be related to a bad power supply. So I broke the printer down and I'm waiting for a replacement PSU.

Also, you should really enable line numbers and checksum in octoprint's config.yaml:
```
feature:
  alwaysSendChecksum: true
  ignoreIdenticalResends: true
  identicalResendsCountdown: 3
```
You will need to set identicalResendsCountdown to BUFSIZE -1.

What this does is enable checksums, so that if there is an error in the command, octoprint can resend the command. Works pretty well.

Beherith, 2018-02-28:
So after all this debugging, and making sure commands are resent on errors, the printer still pauses at least 8 times during a 2 hour print. 
Some commands arrive and fail the checksum check, those are resent but the position (layer height) of the pauses does not correlate with resent commands.

Beherith, 2018-02-28:
SOLVED: I did a quick check of which commands where taking more than 2 seconds to process, and it was a different command every 5 minutes. The culprit was SD_SETTINGS, I dont even know why writing settings to the sd card mid print is a halfway good idea, I just enabled it cause the Due has no eeprom :(

johannes-otto, 2018-03-01:
@Beherith 
Did you go back to 4.3.3? Does everything work there? I just switched to my 32bit combo. Originally it was there sitting and waiting for a rainy day, but on the weekend my 8bit board blew. 
So I have no experiences with previous versions.

Beherith, 2018-03-02:
@johannes-otto My problems were two fold:
1. The SDSettings bug - disable sdsettings in ALL config files! 
2. Communication errors - force octoprint to send checksum
octoprint's config.yaml:
feature:
  alwaysSendChecksum: true
  ignoreIdenticalResends: true
  identicalResendsCountdown: 3
Did you try number 2?

johannes-otto, 2018-03-02:
@Beherith 
1. I disabled SD anyway, due to a bug with my AZSMZ
2. I just edited the ~/.octoprint/config.yaml to match your settings

I'll report back

johannes-otto, 2018-03-03:
@Beherith 
First Benchy was ok. Second had the characteristic blob.

Beherith, 2018-03-03:
Pastebun you serial log from octoprint please

johannes-otto, 2018-03-03:
I now went down with the speed. It seems to work again. I enabled serial logging. I'll put the log of the next print.

johannes-otto, 2018-03-04:
I don't have any hickups anymore. I'm printing on 40mm/s. 
Segments_per_second = 200
Microstepping: 32
Baudrate: 115200

I attached the log. I'll leave logging enabled. When my next print fails, I'll upload it.

[serial.log.2018-03-03_19-01-05.txt](https://github.com/MKFirmware/MK4duo/files/1778743/serial.log.2018-03-03_19-01-05.txt)

Beherith, 2018-03-05:
Ok

ozzysv, 2018-03-09:
The same problem, I print from the sd card. So the impression is that the exchange failed with the card. the rate dropped but it did not have any effect. now 
I'll go to the old version 4.3.29 dev, which is the most stable in me. 

After the transition, I'll write whether the problem is persisted

johannes-otto, 2018-03-10:
@Beherith 
I bumped the speed to 60mm/s print- and 100mm/s travel speed. I get hickups at nearly every second layer. I attached the log.
[serial.log.txt](https://github.com/MKFirmware/MK4duo/files/1799308/serial.log.txt)

johannes-otto, 2018-03-10:
I hope this can be solved somehow. Currently I need to print slower with my shiny 32bit board than with my crappy trigorilla.

seems these problems don't exist with 8bit?

ozzysv, 2018-03-10:
> seems these problems don't exist with 8bit?

as my practice showed - 8 bit versions are much more stable than version 32. 
I think this is the consequences of migration to  the library HAL ((

MagoKimbra, 2018-03-10:
But all this when you print from serial?
I only print from serial with my 32 bit card and I have no slowing down, the two uses the hardware serial libraries contained in the arduino variants for the two.
Or the problem is communicating with Octroprint and / or Astroprint. It could be some automatic function for the two like the auto sending of the temperature without M105 request, because modern hosts know how to see if the firmware has the auto answer and does not send M105 every n seconds. But if instead it continues to send M105 it would have multiple responses both from the self-reply and from the answer itself of M105, and this could give problems.

MagoKimbra, 2018-03-10:
How much have you set BLOCK_BUFFER_SIZE and BUFSIZE?
i have:
#define BLOCK_BUFFER_SIZE 32
#define MAX_CMD_SIZE 96
#define BUFSIZE 8

johannes-otto, 2018-03-10:
@MagoKimbra 
#define BLOCK_BUFFER_SIZE 32
#define MAX_CMD_SIZE 96
#define BUFSIZE 8

I'm using octoprint, since the sdcard on my azsmz lcd doesn't work. Could it be related to my slicer? I'm using Simplify3D. I had no problems on 8bit with it.