MKFirmware/MK4duo #404: Is it possible to define a different pin for the bed

onekk, 2018-02-22:
Sorry if there is not the right place and TIA for the help.

The full machine I'm assemblying is a P3Steel (Toolson Edition).

I have a RAMPS 1.4 (Chinese clone) and want to use a different pin (maybe d5 or d4) to control an external mosfet to drive the Heated Bed.
I have a LCD that work in Marlin with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

In my intention is to have a RAMPS RRD (fan extender) that has two output driven by the D6 and D11 pin that in my intentions would be used to trigger the fans to cool the extruders (D6) and the fan to cool the piece (D11)

The riser of the RRD could be used to bring the D4 or D5 pin to the additional sevenswitch board to control the Heated Bed. 

My intentions are to have two hotend (for now i will start with 1 hotend).

I think that the RAMPS1.4 HHF seems to me a good starting point for the "base" configuration.

And a I didn't want to move pins across the actual one hotend and the future two hotend configuration

Now I have some question:

1) how to control the Heated Bed with the D4 or the D5 pin (assuming they are both good for PWM through the additional board)

2) if I tell MK4duo that i have a dual hotend but only one is wired, there is a mean to control it with the panel (I could recompile another firmware when i install the second hotend, but just in case...)

3) The fan controlled by the HHF could be Fan that cools the ramps? (controlled when the steppers are running)

If this is not the right place, plese tell me  where I can ask in a correct way.

Regards

Carlo D.

MagoKimbra, 2018-02-22:
If you activate HHF you have the outputs available for two hotends and a fan. To activate the second Hotend immediately, you must immediately set two extruders ....
In 4.3.5 if you do not use it there is no problem.
The Fan is the one to cool the PLA, to ventilate the ramps you will have to choose another exit that then with an external circuit you will have to bring 12v.
The second fan is the configurator enablement for auto control of the board (controller)

onekk, 2018-02-22:
So if i configure HHF with 4.3.5 I have to not change the configuration for the hotend, ok for the 2) 
the Fan in the HHF is the fan that cool the PLA, if i have guess right.

For the HotBed control with the external mosfet, (my question number 1) how to control it say with pin D4.

For the ramps, maybe using (D6 or D11) (auto control ?) where i can set the values?

Another little question? it is possible to compile with platformio?

Many Thanks and Regards

MagoKimbra, 2018-02-22:
All configuration with configurator online
www.malinkimbra.it

onekk, 2018-02-22:
No luck with configuration file for the heated bed, if i manually put 

`#define ORIG_HEATER_BED_PIN   4`
in Configuration_Overall.h

it will make the pin4 to became the heated bed pin?

MagoKimbra, 2018-02-22:
No if you do not use configurator, you must open configuration_pins.h and chage 
#define HEATER_BED PIN 4

onekk, 2018-02-22:
Ok, I'm puzzled, in the header of configuration_overall,h is specified that 

`/**
 * Configuration_Overall.h
 * Here you can define all your custom settings and they will overwrite configurations in the main configuration files.
 */
`

Your former response means that the ORIG_???? pins are not overriden by this file?

Just in case some other settings are to be changed that contain ORIGIN_.

Plus I have read some tutorial for setting the CONTROLLERFAN_PIN for Marlin, but I didn't find anything in the MK4duo configs, (I did a grep -R on the main sources cloned from github)

For custom setting what is better a centralized way like in Configuration_Overall is a good way but it could be used for manual tuning?

Regards

MagoKimbra, 2018-02-22:
No overall is file for configuration for example you download the firmware anh change only baudrate without changing the standard values in configuration.h, you can add #define BAUDRATE 115200 in configuration_overall.h, the same for other values.
For pin the file is configuration_pins. The OTIG_ pins is defined in file of the board, for example if you have RAMPS_HFB the file in src/boards/33.h, in this file defined ORIG_ pins...

onekk, 2018-02-23:
OK many thanks, for the controller_fan I have to dig and eventually open another issue this one is off-topic.

closed 2018-02-23 by onekk