MKFirmware/MK4duo #406: Bug: at start a message "Need Tune PID" No bet PID tuning menu

Hans007a, 2018-02-23:
After change to the version 4.3.5 i get at start a message "Need Tune PID"
even though I set the right PID in the configuration_temperature.h.

And I set PID for the heater and bed and this worked for the v 4.3.4 without problems.
 In the menu on the display (RRGD 12864) I have only a PID heater set menu, but nothing for the bed!

My printer are cartesian und core xy, both with due & ramps_fd v1

MagoKimbra, 2018-02-23:
Fix!

closed 2018-02-23 by MagoKimbra

boubouexpress, 2018-03-04:
After change to the version 4.3.5 i get at start a message "Need Tune PID"

boubouexpress, 2018-03-04:
/**
 * MK4duo Firmware for 3D Printer, Laser and CNC
 *
 * Based on Marlin, Sprinter and grbl
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 * Copyright (C) 2013 Alberto Cotronei @MagoKimbra
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

/**
 * Configuration_Overall.h
 * Here you can define all your custom settings and they will overwrite configurations in the main configuration files.
 */

/********************************
 * Firmware Version V4.3.5      *
 ********************************/

#define CONFIGURATION_OVERALL

/***********************
 * Configuration_Basic *
 ***********************/
#define SERIAL_PORT 0
#define BAUDRATE 115200
#define STRING_CONFIG_H_AUTHOR "(R2D2, CR-10S)"
#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
#define KILL_METHOD 0
#define NO_TIMEOUTS 1000
//#define ADVANCED_OK
//#define EMERGENCY_PARSER
//#define FASTER_GCODE_PARSER
#define HOST_KEEPALIVE_FEATURE
#define DEFAULT_KEEPALIVE_INTERVAL 2
#define BUSY_WHILE_HEATING
#define MOTHERBOARD BOARD_RAMPS_13_HFB
#define MECHANISM MECH_CARTESIAN
//#define MECHANISM MECH_COREXY
//#define MECHANISM MECH_COREYX
//#define MECHANISM MECH_COREXZ
//#define MECHANISM MECH_COREZX
//#define MECHANISM MECH_COREYZ
//#define MECHANISM MECH_COREZY
//#define MECHANISM MECH_DELTA
//#define MECHANISM MECH_MORGAN_SCARA
//#define MECHANISM MECH_MAKERARM_SCARA
//#define MECHANISM MECH_MUVE3D
#define POWER_SUPPLY 0
#define PS_DEFAULT_OFF false
#define DELAY_AFTER_POWER_ON 5
#define POWER_TIMEOUT 30
#define EXTRUDERS 1
#define DRIVER_EXTRUDERS 1

/*****************************
 * Configuration_Temperature *
 *****************************/
//#define TEMPERATURE_UNITS_SUPPORT
[... 1281 lines cut ...]
"filrunoutpin": "NoPin",
"filrunoutpin1": "NoPin",
"filrunoutpin2": "NoPin",
"filrunoutpin3": "NoPin",
"filrunoutpin4": "NoPin",
"filrunoutpin5": "NoPin",
"laserpwrpin": "ORIG_LASER_PWR_PIN",
"laserpwmpin": "ORIG_LASER_PWM_PIN",
"laserperipheralspin": "NoPin",
"laserperipheralsstatuspin": "NoPin",
"cncrouterpin": "NoPin",
"powerconsumptionpin": "NoPin",
"doorpin": "NoPin",
"powercheckpin": "NoPin",
"caselightpin": "NoPin",
"rgbledRpin": "NoPin",
"rgbledGpin": "NoPin",
"rgbledBpin": "NoPin",
"rgbledWpin": "NoPin",
"neopixelpin": "NoPin",
"dhtdatapin": "NoPin",
"END_DATA": 0
}
========== End configuration string ==========
*/

MagoKimbra, 2018-03-04:
It's normal for force pid tune and save in eeprom the new value...

boubouexpress, 2018-03-04:
?

MagoKimbra, 2018-03-04:
To force the pid tuning I put a variable that at the beginning is false and after tuning or after entering the pid values with 301 it is put to true and saved in eeprom, at which point they will no longer be requested.