MKFirmware/MK4duo #407: Bug: CoreXY nozzle drive after stop a print into a reverse y direction

Hans007a, 2018-02-23:
My second printer CoreXY drive the nozzle after stop a print (from menu!) into a reverse y direction and crash into case... when the job is normal finished is all ok.

printer: corexy, due + ramps-fd v1 + rrgd

MagoKimbra, 2018-02-24:
When you stop, the firmware stops the printer and saves the position to be able to start again with the file restart.gocde. To do this block the interrupts, reset the buffers, because it must be instantaneous even for use with the voltage control pin. All this, however, if you use the printer immediately after the stop creates a bit of problems like what you indicate. For now, after the printing stops, a reset is to switch off the printer.

Hans007a, 2018-02-25:
btw. after print is completed the nozzle drive to x=0 correctly and stay at the y coordinate

MagoKimbra, 2018-02-25:
No, parking feature is used only in pause not in STOP. Stop clear all buffer, all interrupt and move only Z.

Hans007a, 2018-02-26:
then it's a bug, but pressing Stop Print move X and Y

OlegRRR, 2018-02-27:
I have the same problem. Issue #391. Version 4.3.3 did not have this problem. 4.3.4 and 4.3.5 have it.
If You press "Pause print" first, then "Stop print", the printer stops correctly. While I use so...

closed 2018-04-29 by Hans007a