MKFirmware/MK4duo #414: Support for polar printers?

vanbins, 2018-03-01:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Tell us here what happened: don't miss any valuable detail, be clear and concise, don't use UPPERCASE or abbreviations which can make understanding what you wrote quite difficult.

### Compile errors
If you have problems compiling MK4duo, report here the error messages Arduino IDE gives you.

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version
MK4duo's versions that have this issue.

### My setup
If you used the online configurator, copy the content of `Configuration_Overall.h` and `Configuration_Pins.h` in two text files. **Then Drag&Drop them here. DO NOT COPY THE CONTENT DIRECTLY HERE!!!**

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

vanbins, 2018-03-01:
Hi Sorry, I pressed the Enter key a bit too early :/ Anyways I'm working on making a polar printer, and right now the gcode for it is being made manually. The machine is currently using Repetier and it's working fine (still needs a lot of tweaking, but it prints), but it would be great if "regular" gcode could be used, made by a slicer. Converting Cartesian coordinates into polar ones isn't the thing i'm struggling with that much, but converting the speed and extruder feedrate is what is complicated. The printer has some unusual "features" right now. In example, One unit on the Y doesn't stand for 1 degree of movement as most polar systems, but rather for 1 rotation of the turntable.
Could polar kinematics be added to MK4duo? It does say "A single Firmware for all types of printers; Cartesian, Delta, MORGAN SCARA, MAKERARM SCARA, CoreXY, CoreXZ & CoreYZ and their reverse."

MagoKimbra, 2018-03-01:
Yes, you're right, but after some big changes I left SCARA, because unfortunately I have no feedback and no one to give me a hand in the programming of these printers.