MKFirmware/MK4duo #415: Compile error with mk4duo 4.3.5

vabendai, 2018-03-01:
Compile error with mk4duo 4.3.5

`C:\Users\FFISIC~1\AppData\Local\Temp\arduino_build_728711\sketch\src\core\heater\heater.cpp:92:8: error: prototype for 'void Heater::get_pid_output(bool)' does not match any in class 'Heater'`

   `void Heater::get_pid_output(const bool cycle_1s) {
`
        ^

`In file included from C:\Users\FFISIC~1\AppData\Local\Temp\arduino_build_728711\sketch\src\core\heater\../../../MK4duo.h:131:0,`

`                 from C:\Users\FFISIC~1\AppData\Local\Temp\arduino_build_728711\sketch\src\core\heater\heater.cpp:27:`

`C:\Users\FFISIC~1\AppData\Local\Temp\arduino_build_728711\sketch\src\core\heater\../../../src/core/heater/heater.h:102:12: error: candidate is: void Heater::get_pid_output(uint8_t)`
`
       void get_pid_output(const uint8_t cycle_1s);`

`            ^`

Many thanks

[Configuration_Overall.zip](https://github.com/MKFirmware/MK4duo/files/1770981/Configuration_Overall.zip)

onekk, 2018-03-01:
Same error for me, but a prior compilation went fine, maybe different sources with the same firmware version 4.3.5 ?

onekk, 2018-03-02:
UPDATE: loading the previous configuration in online configurator and recompiling the new zip obtained gave no error.

So for me this issue is SOLVED. as 02 March 2018 09:56.

Regards

vabendai, 2018-03-02:
I am also able to compile

closed 2018-03-02 by vabendai