MKFirmware/MK4duo #419: A few remarks about the TMC2208 (and other) driver

MikeSmith78, 2018-03-04:
1. Control of direction rotation through interface is a bad idea. I use TMC2208. UART-interface is slow and unreliable. The result of this is an unstable motion.

\src\core\stepper\stepper_indirection.h
macro X_DIR_INIT, X_DIR_WRITE, X_DIR_READ etc
I think that this algorithm should be made as an option.

2. TRINAMIC driver initialization error.
If EEPROM not used, basic parameters are initialized with the default values in function eeprom.Load_Settings(). But stepper driver is initialized before the configuration variables takes its actual value. And some registers of TMC2208 are initialized incorrectly.

I think that the initialization of factory settings should be done earlier.
src/core/printer/printer.cpp
lines 189 and 195

### Firmware version
Master branch, commit 1907

MagoKimbra, 2018-03-04:
Ok thank's for info...

MagoKimbra, 2018-03-04:
Try now a move stepper.init post load eeprom..

MikeSmith78, 2018-03-05:
Yes, it solved this problem

closed 2018-03-05 by MikeSmith78