MKFirmware/MK4duo #420: Bug: wrong movement at homing on CoreXY

Hans007a, 2018-03-04:
After the last update (Fix #418) my printer moves wrong:

I press the home function and after a littlepause
Home X : moving to x=0 and y -> max
Home Y : moving to y =0 and y -> max

and I have to press reset to stop it!

My printer is TronXY X5S CoreXY with BLTouch, Due + RAMPS-FD v1

MagoKimbra, 2018-03-04:
I delete only:
```
#if MINIMUM_STEPPER_PULSE == 0
  #undef MINIMUM_STEPPER_PULSE
  #define MINIMUM_STEPPER_PULSE 1
#endif
```

Hans007a, 2018-03-04:
I check my sources for the changes

Hans007a, 2018-03-08:
last commit (1920) again wrong moving. back to the the working code...

iosonopersia, 2018-03-08:
I examined those two "Clear code" commits: in my opinion the error cannot be caused by c41992d15abad6d11e7b9f37bf2609baa05137d4 , while as far as fb25a6f42f70dcfd636b93b34ab7780a629d2ce4 is concerned I believe the only relevant changes that may have caused problems were made inside core_mechanics and stepper classes

iosonopersia, 2018-03-08:
@Hans007a please, download MK4duo from here (https://github.com/MKFirmware/MK4duo/tree/c41992d15abad6d11e7b9f37bf2609baa05137d4) and tell us if it still worked correctly back then. This way we can know with absolute certainty which one of the two commits is broken.

Hans007a, 2018-03-08:
ok, I do this tomorrow and write my results here :)

Hans007a, 2018-03-08:
I checked it just now,and it works :)

iosonopersia, 2018-03-08:
Thanks! That's a very valuable info! We now know that the problem is caused by fb25a6f ... but I don't see how could it be possible that those changes are correlated to the new uncorrect behaviour of your printer...

iosonopersia, 2018-03-08:
@Hans007a , is FWRETRACT enabled in your configuration files?

Hans007a, 2018-03-09:
Hello @iosonopersia,

no, FWRETRACT is not enabled in the feature settings.

Hans007a, 2018-03-09:
btw. I print from SD Card, maybe there is the bug?

iosonopersia, 2018-03-09:
@Hans007a, sorry to bother you again, could you please try again with this version? https://github.com/MKFirmware/MK4duo/tree/fb25a6f42f70dcfd636b93b34ab7780a629d2ce4

Hans007a, 2018-03-09:
It's OK :) This helps me too!

I can say, this version works too!

Hans007a, 2018-03-13:
it looks like this issue are gone. I close this thread

closed 2018-03-13 by Hans007a