MKFirmware/MK4duo #421: Planner error

MikeSmith78, 2018-03-05:
### Issue description
I found a problem in the planner algorithm. Error occurs when:
- all motors are stopped, X = 0; 
- one-axis movement command is sent; for example, **G1 X50**;
- the planner correctly prepares the first trapezoid block (zero start and stop speed, acceleration and deceleration stages are present);
- the x axis starts to move;
- then two commands are sent: **G1 X100** and **G1 X150**;
- the planner will not correctly process the second command: the first segment will end with a stop motion, but the second segment will start at full speed. In this case, the synchronization lost of the stepping motor is possible.

This picture will clarify the problem:
[image]

- time 1-2 - first segment acceleration:
- time 2-3 - first segment move at required speed;
- time 3-4 - first segment deceleration and full stop;
- **point 4 - start move at required speed without prior acceleration;**
- time 4-5 - move to point X100 segment;
- time 5-7 - third segment (no problems);
- time 7-10 - return to the starting point X=0 (no problems).

My test program:
```
M201 X500 Y500
G1 X50 F6000
G4 P100
G1 X100
G1 X150
G1 X50
```

Probably this problem occurs only with a linear motion in one direction.

### Firmware version
Master branch, commit 1907

### My hardware
CoreXY

MagoKimbra, 2018-03-05:
What oscilloscope do you use?
Beautiful ...
With which commit did you try the last or an older version?

MikeSmith78, 2018-03-05:
No, this is not oscilloscope. This is my own debug interface. Commit 1907. But I think that this is quite an old problem...

MikeSmith78, 2018-03-05:
Sorry, this is the correct gcode
```
M201 X500 Y500
G1 X50 F6000
G4 P100
G1 X100
G1 X150
G1 X0
```

MagoKimbra, 2018-03-05:
Have you tried with the last one? I can know the model of your interface.

MikeSmith78, 2018-03-05:
Yes, X0->X50->X100-X150-X0.
Did you mean my debug interface? This is a combination of its own control protocol, a set of debugging commands, a sub-program of the lower level, my high-level program with a graphical interface on the computer ...
If you need low-level debugging, I think you can use Saleae Logic Analyzer with the required further processing of raw results.

MagoKimbra, 2018-03-05:
It was the one I wanted to buy the saleae logic.
I wanted to know if you're trying with the latest version that's online?

MikeSmith78, 2018-03-06:
Yes, I'm using the latest version.
Now I'm studying this problem. And a later I will provide additional information.

MikeSmith78, 2018-03-06:
Sorry again. Using a pause **G4 P100** changes the result of the planner.
Here is the code that gives the picture above:
```
;G28; do it once
M201 X100 Y100
G1 X50 F6000
G1 X100
G1 X150
G1 X0
```
Please check it on you printer.

MikeSmith78, 2018-03-06:
Contents of Planner::block_buffer step by step:

[image]

- initial state: all motors are off; planner buffer is empty;
- **Step1**: first command G1 **X50 F6000** is sent; planner make first block correctly; and stores its nominal speed (variable **previous_nominal_speed**);
- this first block is blocked (by setting flag **BLOCK_BIT_BUSY**) and the motors start working;
- **Step2**: second command **G1 X100** is sent; the planner prepares a second block; acceleration and deceleration stages are ok; but interval t3-t5 does not changed, because function Planner::reverse_pass worked with more than 2 segments;
- **Step3**: start processing of the third segment **G1 X150**; this segment move to planner buffer;
- **Step4**: last call of function **Planner::recalculate**; unnecessary acceleration and deceleration are removed; but segment 1 does not change because its flag is **BLOCK_BIT_BUSY**. As result - segment 2 start at nominal speed without necessary acceleration.

I think it is necessary to limit value of variables **entry_speed** and **max_entry_speed** if previous block is busy.

MagoKimbra, 2018-03-06:
Good... Now check this and test. if you are news write me... Thank's

MagoKimbra, 2018-03-07:
Please check with last commit master!!!

MikeSmith78, 2018-03-07:
[image]

MagoKimbra, 2018-03-07:
It's Good...

MikeSmith78, 2018-03-07:
Next problem:

[image]

Start point: X=250
```
;G28
M201 X100 Y100
G1 X0
G1 X50 F6000
G1 X100
G1 X150
G1 X200
G1 X250
```
Last move X 200->250 have unnecessary pause.

MikeSmith78, 2018-03-15:
Well, let's consider this not a bug, but a feature.

This pause (move 200->250) is formed because function **Planner::reverse_pass()**  does not process the last block. My working solution:

```
void Planner::reverse_pass_kernel(block_t* const current, const float exit_speed) {
  if (!current) return;
  float max_entry_speed = current->max_entry_speed;
  if (current->entry_speed != max_entry_speed) {
    current->entry_speed = (TEST(current->flag, BLOCK_BIT_NOMINAL_LENGTH) || max_entry_speed <= exit_speed)
      ? max_entry_speed
      : min(max_entry_speed, max_allowable_speed(-current->acceleration, exit_speed, current->millimeters));
    SBI(current->flag, BLOCK_BIT_RECALCULATE);
  }
}

void Planner::reverse_pass() {
  if (movesplanned() > 2) {
    const uint8_t endnr = BLOCK_MOD(block_buffer_tail + 1);
    uint8_t blocknr     = prev_block_index(block_buffer_head);
    block_t* current    = &block_buffer[blocknr];
    float final_speed = MINIMUM_PLANNER_SPEED;
    do {
      if (TEST(current->flag, BLOCK_BIT_START_FROM_FULL_HALT))
        break;
      reverse_pass_kernel(current, final_speed);
      final_speed = current->entry_speed;
      blocknr = prev_block_index(blocknr);
      current = &block_buffer[blocknr];
    } while (blocknr != endnr);
  }
}
```

MikeSmith78, 2018-05-20:
The planner is rewritten, and this solution is no longer relevant.

closed 2018-05-20 by MikeSmith78