MKFirmware/MK4duo #423: Can't compile the las commit

Hans007a, 2018-03-07:
Here my error log:

sketch\src\feature\printcounter\printcounter.cpp:27:28: error: definition of implicitly-declared 'constexpr PrintCounter::PrintCounter()'

 PrintCounter::PrintCounter(): super() {

                            ^

sketch\src\feature\printcounter\printcounter.cpp: In static member function 'static millis_t PrintCounter::deltaDuration()':

sketch\src\feature\printcounter\printcounter.cpp:36:18: error: 'this' is unavailable for static member functions

   millis_t tmp = this->lastDuration;

                  ^

sketch\src\feature\printcounter\printcounter.cpp:37:3: error: 'this' is unavailable for static member functions

   this->lastDuration = this->duration();

   ^

sketch\src\feature\printcounter\printcounter.cpp:37:24: error: 'this' is unavailable for static member functions

   this->lastDuration = this->duration();

                        ^

sketch\src\feature\printcounter\printcounter.cpp:38:10: error: 'this' is unavailable for static member functions

   return this->lastDuration - tmp;

          ^

sketch\src\feature\printcounter\printcounter.cpp: In static member function 'static void PrintCounter::initStats()':

sketch\src\feature\printcounter\printcounter.cpp:46:3: error: 'this' is unavailable for static member functions

   this->data = { 0, 0, 0, 0, 0.0 };

   ^

sketch\src\feature\printcounter\printcounter.cpp: In static member function 'static void PrintCounter::saveStats()':

sketch\src\feature\printcounter\printcounter.cpp:68:8: error: 'this' is unavailable for static member functions

   if (!this->loaded) return;

        ^

In file included from sketch\src\feature\printcounter\../../../src/HAL/HAL.h:40:0,

                 from sketch\src\feature\printcounter\../../../MK4duo.h:117,

                 from sketch\src\feature\printcounter\printcounter.cpp:23:

sketch\src\feature\printcounter\printcounter.cpp: In static member function 'static void PrintCounter::showStats()':

sketch\src\feature\printcounter\printcounter.cpp:80:24: error: 'this' is unavailable for static member functions

   SERIAL_MV("Total: ", this->data.totalPrints);

                        ^

sketch\src\feature\printcounter\../../../src/HAL/HAL_DUE/communication.h:104:57: note: in definition of macro 'SERIAL_VAL'

 #define SERIAL_VAL(val, ...)                (Com::print(val, ## __VA_ARGS__))

                                                         ^

[... 220 lines cut ...]
     ^

sketch\src\feature\printcounter\printcounter.cpp:165:5: error: 'this' is unavailable for static member functions

     this->data.printTime += this->deltaDuration();

     ^

sketch\src\feature\printcounter\printcounter.cpp:165:29: error: 'this' is unavailable for static member functions

     this->data.printTime += this->deltaDuration();

                             ^

sketch\src\feature\printcounter\printcounter.cpp:166:5: error: 'this' is unavailable for static member functions

     this->saveStats();

     ^

sketch\src\feature\printcounter\printcounter.cpp: In static member function 'static void PrintCounter::reset()':

sketch\src\feature\printcounter\printcounter.cpp:178:3: error: 'this' is unavailable for static member functions

   this->lastDuration = 0;

MagoKimbra, 2018-03-07:
You have copied the folder to the old one, printocounter is no longer in the feature folder so it can not be that error, most likely you still have the old folder.

Hans007a, 2018-03-08:
OK, I will check that!

closed 2018-03-08 by Hans007a