MKFirmware/MK4duo #425: ultratronics & TMC2130 compile error

silversat, 2018-03-10:
### Compile errors

In file included from sketch\src\core\commands\gcode/gcode.h:64:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/config/m906.h: In function 'void gcode_M906()':

sketch\src\core\commands\gcode/config/m906.h:62:57: error: 'extended_axis_codes' was not declared in this scope

       if (values[P##_AXIS]) tmc_set_current(stepper##Q, extended_axis_codes[TMC_##Q], values[P##_AXIS]); \

                                                         ^

sketch\src\core\commands\gcode/config/m906.h:66:7: note: in expansion of macro 'TMC_SET_GET_CURRENT'

       TMC_SET_GET_CURRENT(X,X);

       ^

sketch\src\core\commands\gcode/config/m906.h:63:40: error: 'extended_axis_codes' was not declared in this scope

       else tmc_get_current(stepper##Q, extended_axis_codes[TMC_##Q]); } while(0)

                                        ^

sketch\src\core\commands\gcode/config/m906.h:66:7: note: in expansion of macro 'TMC_SET_GET_CURRENT'

       TMC_SET_GET_CURRENT(X,X);

       ^

sketch\src\core\commands\gcode/config/m906.h:62:57: error: 'extended_axis_codes' was not declared in this scope

       if (values[P##_AXIS]) tmc_set_current(stepper##Q, extended_axis_codes[TMC_##Q], values[P##_AXIS]); \

                                                         ^

sketch\src\core\commands\gcode/config/m906.h:72:7: note: in expansion of macro 'TMC_SET_GET_CURRENT'

       TMC_SET_GET_CURRENT(Y,Y);

       ^

sketch\src\core\commands\gcode/config/m906.h:63:40: error: 'extended_axis_codes' was not declared in this scope

       else tmc_get_current(stepper##Q, extended_axis_codes[TMC_##Q]); } while(0)

                                        ^

sketch\src\core\commands\gcode/config/m906.h:72:7: note: in expansion of macro 'TMC_SET_GET_CURRENT'

       TMC_SET_GET_CURRENT(Y,Y);

       ^

sketch\src\core\commands\gcode/config/m906.h:62:57: error: 'extended_axis_codes' was not declared in this scope

       if (values[P##_AXIS]) tmc_set_current(stepper##Q, extended_axis_codes[TMC_##Q], values[P##_AXIS]); \

                                                         ^

sketch\src\core\commands\gcode/config/m906.h:78:7: note: in expansion of macro 'TMC_SET_GET_CURRENT'

       TMC_SET_GET_CURRENT(Z,Z);

       ^

sketch\src\core\commands\gcode/config/m906.h:63:40: error: 'extended_axis_codes' was not declared in this scope

       else tmc_get_current(stepper##Q, extended_axis_codes[TMC_##Q]); } while(0)

                                        ^

sketch\src\core\commands\gcode/config/m906.h:78:7: note: in expansion of macro 'TMC_SET_GET_CURRENT'

       TMC_SET_GET_CURRENT(Z,Z);

       ^

sketch\src\core\commands\gcode/config/m906.h:62:57: error: 'extended_axis_codes' was not declared in this scope

       if (values[P##_AXIS]) tmc_set_current(stepper##Q, extended_axis_codes[TMC_##Q], values[P##_AXIS]); \

                                                         ^

sketch\src\core\commands\gcode/config/m906.h:84:7: note: in expansion of macro 'TMC_SET_GET_CURRENT'

       TMC_SET_GET_CURRENT(E,E0);

       ^

sketch\src\core\commands\gcode/config/m906.h:63:40: error: 'extended_axis_codes' was not declared in this scope

       else tmc_get_current(stepper##Q, extended_axis_codes[TMC_##Q]); } while(0)

                                        ^

sketch\src\core\commands\gcode/config/m906.h:84:7: note: in expansion of macro 'TMC_SET_GET_CURRENT'

       TMC_SET_GET_CURRENT(E,E0);

       ^

In file included from sketch\src\core\commands\gcode/gcode.h:116:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/feature/m911_m915.h: In function 'void gcode_M911()':

sketch\src\core\commands\gcode/feature/m911_m915.h:39:33: error: 'extended_axis_codes' was not declared in this scope

       tmc_report_otpw(stepperX, extended_axis_codes[TMC_X]);

                                 ^

In file included from sketch\src\core\commands\gcode/gcode.h:116:0,

                 from sketch\src\core\commands\commands.cpp:30:

sketch\src\core\commands\gcode/feature/m911_m915.h: In function 'void gcode_M912()':

sketch\src\core\commands\gcode/feature/m911_m915.h:61:56: error: 'extended_axis_codes' was not declared in this scope

       if (clearX || clearAll) tmc_clear_otpw(stepperX, extended_axis_codes[TMC_X]);

                                                        ^

sketch\src\core\commands\gcode/feature/m911_m915.h:68:56: error: 'extended_axis_codes' was not declared in this scope

       if (clearY || clearAll) tmc_clear_otpw(stepperY, extended_axis_codes[TMC_Y]);

                                                        ^

sketch\src\core\commands\gcode/feature/m911_m915.h:75:56: error: 'extended_axis_codes' was not declared in this scope

       if (clearZ || clearAll) tmc_clear_otpw(stepperZ, extended_axis_codes[TMC_Z]);

                                                        ^

sketch\src\core\commands\gcode/feature/m911_m915.h:82:57: error: 'extended_axis_codes' was not declared in this scope

       if (clearE || clearAll) tmc_clear_otpw(stepperE0, extended_axis_codes[TMC_E0]);

                                                         ^

exit status 1
Errore durante la compilazione per la scheda Arduino Due (Programming Port).

### Firmware version
MK4duo 4.3.5

### My setup
#define HAVE_TMC2130
&
  #define X_IS_TMC2130
  #define Y_IS_TMC2130
  #define Z_IS_TMC2130
  #define E0_IS_TMC2130

closed 2018-03-10 by MagoKimbra

silversat, 2018-03-10:
hi magokimbra, you have the same issue into m911_m915.h header file
bye

MagoKimbra, 2018-03-10:
Sorry... fixed it...

commit 2d7f35ea (2018-03-10): fix #425