MKFirmware/MK4duo #426: ABL bilinear anything wrong

Hans007a, 2018-03-10:
My bed is 300x300mm and I have BLTouch to bed leveling.

Now I use linear bed leveling, because bilinear (that should be better for my big bed) provides wrong results and the nozzle is going into the bed at one edge (left, left-behind). linear is not perfect for my bed, but the nozzle stay up the bed.

maybe any sign in the calculation of the bilinear grid?

printer that not working: corexy, duo+ramps-fd v1 + eeprom

MagoKimbra, 2018-03-10:
Do you active ABL when start printing with M420 S1?

Hans007a, 2018-03-10:
I thing, wenn I print from sd card and make G29 from gcode, then ABL is active for the print or not?

MagoKimbra, 2018-03-10:
If you do G28 and g29 every time you ok it's ok so it's already active!

Hans007a, 2018-03-11:
I know, I can see it at the z motors by printing.

I will check the UBL. maybe this will be better

Hans007a, 2018-03-11:
I saw, You changed many places in the code and I will compile the newest version and hope that the wrong moves are gone and I can try the UBL...

Hans007a, 2018-03-14:
I don't know why, but it works now better, after I switch back from not working UBL to ABL.
I close this issue.

closed 2018-03-14 by Hans007a