MKFirmware/MK4duo #427: Buffer overflow in gcode_M106()

MikeSmith78, 2018-03-10:
### Issue description
**file** \MK4duo\src\core\commands\gcode\control\m106_m107.h
**line** 82-89:
```
char response[50];
sprintf_P(response, PSTR("Fan: %i pin: %i frequency: %uHz min: %i inverted: %s"),
            (int)f,
            (int)fan->pin,
            (uint16_t)fan->freq,
            (int)fan->min_Speed,
            (fan->isHWInverted()) ? "true" : "false"
);
```
The length of the final line can exceed 50 characters

iosonopersia, 2018-03-11:
This was fixed in commit https://github.com/MKFirmware/MK4duo/commit/d35adc8b8ed59343aa82d5fd19251de377eda12d . 70 chars should be enough to contain the entire string

MagoKimbra, 2018-03-11:
A ok i have lost this issue.. Thank's @iosonopersia ...

closed 2018-03-11 by MagoKimbra