MKFirmware/MK4duo #431: CoreXY at auto home the steps are half of it

Hans007a, 2018-03-13:
At the last commit at the auto home x and x steps are only a half of it.
Manual moving is ok!

My printer: CoreXY + RAMPS-FD v1 + BLTouch + EEPROM, drivers: drv8825

Hans007a, 2018-03-22:
checked the new update and steps per unit at auto home allways wrong! the printer is doing a half of normal steps...

back to my old working version

MagoKimbra, 2018-03-22:
???

Hans007a, 2018-03-22:
I try to exxplain:
when I do auto home (from lcd) the printer move very slow. it looks like he makesnot the normal steps.
I set the safe z homing at the middle of the bed, but the printer move to a quarter of the bed to made z homing...

Hans007a, 2018-03-23:
After last commit from today (23.03.2018) nothing changed. Bug is still present. Back to the last working version.

btw. I changed all source files ald leave only my settings header files for test...

pirogeth, 2018-03-27:
what are your microsteps set at?

Hans007a, 2018-03-28:
I know, my steps are correct, because:
1. manually moving axes (from LCD) works normal
2. old firmware works correct too

I think, its anything in G28 and G29 codes for CoreXY

Hans007a, 2018-03-31:
I checked the newest version from today (31.03.2018) with my cartesian printer and it works there, but on my coreXY NOT! (with the old version is all OK on my coreXY)
I have the same RAMPS-FD with due board, dvr8825 for the axis and bltouch on both, and the same settings except cartesian for one and corexy for the second printer.

On both I have set this:

```
// Change if Z babysteps should go the other way
#define BABYSTEP_INVERT_Z false
// Babysteps are very small. Increase for faster motion.
#define BABYSTEP_MULTIPLICATOR 50
// Enable to combine M851 and Babystepping
#define BABYSTEP_ZPROBE_OFFSET
// Double-click on the Status Screen for Z Babystepping.
//#define DOUBLECLICK_FOR_Z_BABYSTEPPING
// Maximum interval between clicks, in milliseconds.
// Note: Extra time may be added to mitigate controller latency.
#define DOUBLECLICK_MAX_INTERVAL 1250

// Enable graphical overlay on Z-offset editor
#define BABYSTEP_ZPROBE_GFX_OVERLAY
// Reverses the direction of the CW/CCW indicators 
//#define BABYSTEP_ZPROBE_GFX_REVERSE
```
in config_feature.h
maybe is there the bug?

closed 2018-04-29 by Hans007a