MKFirmware/MK4duo #432: Improvement for POWER.CPP

axl12ap, 2018-03-14:
My proposal is to change a line inside POWER.CPP at line 47 where we can read:

`else if (ELAPSED(ms, lastPowerOn + (POWER_TIMEOUT) * 1000UL)) {`

and replace with

`else if ((POWER_TIMEOUT>0) && (ELAPSED(ms, lastPowerOn + (POWER_TIMEOUT) * 1000UL))) {`

because if I change POWER_TIMEOUT to 0 I'd like to disable the feature to power off the printer @ timeout.

Thanks,
Alessandro

iosonopersia, 2018-03-16:
This change was implemented by @MagoKimbra in https://github.com/MKFirmware/MK4duo/commit/a9c96e2d39b5509a9a67e40e4a2be1a355a3be9f

Thank you very much for the suggestion!
This issue can now be closed.

iosonopersia, 2018-08-01:
This old issue was solved 5 months ago, it can be closed @MagoKimbra

closed 2018-08-01 by MagoKimbra