MKFirmware/MK4duo #443: nextion Display with Radds not running

Koobatz, 2018-03-22:
Hello,
With the current firmware 4.3.5, the Nextions display does not work in conjunction with a RADDS board. Display is a 3.5 " Nextion basic display.
I see only in pins folder (1401.h) that the RX & TX pin is not declared, is this normal?
best regards

iworld4us, 2018-03-23:
com port config only in   **Configuration_Overall.h**

Koobatz, 2018-03-23:
thanks for help!
in the Configuration_Overall.h can i only define the serial Port (#define NEXTION_SERIAL 1)
i conect the nextion display on my radds to JP11 and this pins go to the SAM3E on pin 18 & 19.
this is the pins for RX and TX communication. ( http://www.dr-henschke.de/RADDS_27_sch.pdf )

How can I see which serial port is going to which pins in the configuration?

MagoKimbra, 2018-03-23:
Ok Serial 1 is pin 18 & 19. There are no serial pins configured. The hardware is that of the processor.

Koobatz, 2018-03-23:
Ok I understand. then the display works in conjunction with radds and due not :(

MagoKimbra, 2018-03-23:
Yes works. I have ultratronics and alligator with arduino due and work perfectly..
Do you try change tx with rx and the other way around.

Koobatz, 2018-03-24:
ohh man, i switch the tx and rx pin and now runs the display  -.-
please excuse me for making an entry here an thanks for help.

closed 2018-03-31 by Koobatz