MKFirmware/MK4duo #449: HAL_Due.ccp - "PIN_STATUS_PWM" Compiler Error

Koobatz, 2018-03-31:
Hello,
I have the latest firmware with the online configurator created, unfortunately I always get the following error message when compiling and just do not know how to continue. Google could not help me with that, i would be grateful if you could help me. 
I kept the following instructions : https://github.com/MKFirmware/MK4duo/blob/master/Documentation/Compilation.md

### Compile errors
=============
sketch\src\HAL\HAL_DUE\HAL_Due.cpp: In static member function 'static void HAL::analogWrite(pin_t, uint32_t, uint16_t)':
sketch\src\HAL\HAL_DUE\HAL_Due.cpp:415:10: error: 'g_pinStatus' was not declared in this scope
     if ((g_pinStatus[pin] & 0xF) != PIN_STATUS_PWM) {
          ^
sketch\src\HAL\HAL_DUE\HAL_Due.cpp:415:37: error: 'PIN_STATUS_PWM' was not declared in this scope
     if ((g_pinStatus[pin] & 0xF) != PIN_STATUS_PWM) {
                                     ^
sketch\src\HAL\HAL_DUE\HAL_Due.cpp:488:10: error: 'g_pinStatus' was not declared in this scope
     if ((g_pinStatus[pin] & 0xF) != PIN_STATUS_PWM) {
          ^
sketch\src\HAL\HAL_DUE\HAL_Due.cpp:488:37: error: 'PIN_STATUS_PWM' was not declared in this scope
     if ((g_pinStatus[pin] & 0xF) != PIN_STATUS_PWM) {
                                     ^
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Native USB Port).

### Firmware version
===============
MK4duo 4.3.5
Arduino IDE 1.8.5
install SAM 1.6.6

### My setup
=========
Arduino Due + Radds Board 1.6
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/1865520/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/1865521/Configuration_Pins.txt)

MagoKimbra, 2018-03-31:
g_pinStatus is command in arduino library for Arduino due... 
If it does not find it means that there is something wrong with the two downloaded libraries.

Koobatz, 2018-03-31:
why two libarys? i have only downloaded the SAM 1.6.6 libary via the board administrator.
when i updatet the libary > 1.6.8 can i compile the sketch, this causes other problems?

MagoKimbra, 2018-03-31:
I downloaded firmware with your overall, compiled without any errors

Koobatz, 2018-03-31:
I do not get it! i have now reinstall Arduino 1.8.5 and the SAM 1.6.6 libary, the same problem ...
The Programmer is : AVRISP mk||

Koobatz, 2018-04-01:
in the SAM 1.6.6 libary is "g_pinStatus & PIN_STATUS_PWM" not declared, this can not run!

I have Downloaded the SAM 1.6.8 libary [http://downloads.arduino.cc/cores/sam-1.6.8.tar.bz2](http://downloads.arduino.cc/cores/sam-1.6.8.tar.bz2) and copy the files from \sam\cores\arduino - "Arduino.h & wiring_analog.h & wiring_digital.h" into the 1.6.6 libary, now i can compile this Firmware.

I know this topic is insignificant, but do you really use these SAM 1.6.6 libary?

MagoKimbra, 2018-04-01:
Yes the version downloaded with arduino ide is 1.6.11 for sam library.. Why you have 1.6.6 o 1.6.8?

Koobatz, 2018-04-01:
now I do not understand anything anymore. because here one should use the SAM 1.6.6 libary
[https://github.com/MKFirmware/MK4duo/blob/master/Documentation/Compilation.md](https://github.com/MKFirmware/MK4duo/blob/master/Documentation/Compilation.md)
and since i'm not a programmer, i thought this would have a reason.

do you can customize the Compilation.md for other people?

Koobatz, 2018-04-01:
thanks, i closed this thread and thanks for your work.

closed 2018-04-01 by Koobatz