MKFirmware/MK4duo #451: Improvement on ultralcd.cpp for 3in1 Extruder nozzle

Hans007a, 2018-04-02:
I found a bug in ultralcd.cpp: 
line 2892 the value is an integer and it should be:
`if (encoderPosition !=0) {`

and I made a little improvements for the 3in1 nozzle:
- from line 2934:
```
  #if EXTRUDERS == 1
	void lcd_move_e() { _lcd_move_e(); }
  #elif EXTRUDER
```
- from line 2995:
 ```
 #if EXTRUDERS == 1
	void lcd_move_get_e_amount()            { _lcd_move_distance_menu(E_AXIS, lcd_move_e); }
  #elif EXTRUDERS > 1
```
-from line 3062:
```
#if EXTRUDERS == 1
	  MENU_ITEM(submenu, MSG_MOVE_E, lcd_move_get_e_amount);
    #elif EXTRUDERS > 1
```

this do only this that the first extruder dont shows two times on the display :)

btw. any news about the CoreXY ABL bug?

iosonopersia, 2018-08-25:
@Hans007a see #572

iosonopersia, 2018-08-27:
@Hans007a , see #574

Hans007a, 2018-08-28:
Hello!

I saw this and I hope, its good :)

I have my changes still in ultralcd.cpp and it works for me without any problems.

Regards

Hans

commit 12278cfc (2018-09-13): Merge pull request #574 from iosonopersia/scaraDevelopment

Fix #451
commit f016cbd1 (2018-09-13): Fix #451
closed 2018-09-13 by MagoKimbra