MKFirmware/MK4duo #452: Nextion Enhanced 7" Screen and Graphical Screen Togther 4 Hotends and Heatbed

mohala, 2018-04-07:
Hello,

I am very new to the MK4Duo community and printing but I am making a CoreXY 3D printer and I am planning to involve both graphical Screens From SMART RAMPS 12864 SD Card Screen and Nextion 7" Capacitive Touch Enhanced Screen, (which I have bought already)

I have made sure to break out the TX and RX pins from AUX3 so it can be connected to Nextion 7", but I have an issue on implementing both screens on the website configurator it only allows for one and the 12864 SD card screen is not on the list.

Also, another issue is the nextion 7" only shows 2 Hotends, is it possible to add 2 more hotends to make 4 hotends? And how would I go about doing this?

(I am a total beginner in coding but trying my best to learn, so sorry in advance if I need more help)

Best regards 
Mo

MagoKimbra, 2018-04-08:
The Nextion is a display managed directly by the firmware so you can use it alone as the other displays, you can not put multiple displays at the same time. Putting 4 Hotends on the display means redesigning all the graphics, I'm sorry but this is not possible.

mohala, 2018-04-08:
Thank you for the reply, I understand about the multiple screens thank you for letting me know.
But I did attempt to change the design look to accommodate 4 hotends the only issue is I don't know how to do the coding for getting the information to plug into the area for hotend 3 and 4. Would you be able to help me on this and share to everyone? I have the HMI file but not sure how to post it on github.

[image]

mohala, 2019-06-01:
Hello,

Any update for this suggestion?

Mo

MagoKimbra, 2019-06-01:
I am changing all the programming of the Nextion so as to be independent of the graphics and then it can be modified at will. As soon as everything is ready we can verify everything even with your layout ...